nordicsemi/Android-nRF-Connect #89: nRF connect app not detecting ble devices on Huawei 8.0

junto06, 2018-12-18:
My device Huawei Mate 10 running android 8.0 isn't detecting any ble device with nrf connect. I successfully pair it with another phone and laptop bluetooth and got paired. I rebooted my many times but still it doesn't show any  device.  I have another phone Huawei Mate 9 running android 8.0 that is detecting ble devices. Should I try hard reset/wipe data ?

philips77, 2018-12-18:
Before resetting, I would recommend:
1. Ensure the nRF Connect was given the Location permission. 
2. Ensure the Location service is enabled (in Quick settings).
3. Try a different app, like Ble Scanner or nRF Toolbox -> UART -> Connect.
4. In nRF Connect check "Settings -> Scanner" try setting default values: enable Continuous scanning, set Scanning API to Lollipop, use Low latency mode and disable batching and compatibility mode.

Recently Android requires Fine Location Permission to be granted, not Coarse (https://android-review.googlesource.com/c/platform/packages/apps/Bluetooth/+/848935). I'll release an update soon when more strict permission will be requested. But that is to come in Android Q.

junto06, 2018-12-19:
Hello,
1. I double check, location permission was granted.
2.Location servie is enabled.
3.Its not working Ble Scanner, nRF toolbox. I tried few other apps as well.
4.I have double check, settings are same but its not detecting.

Just now I try to pair with my other phone running android 8 and I was able to pair and transfer file from this phone. See this screenshot for reference

https://drive.google.com/open?id=12pY_T8jNgCrTvSv7Y-4G4R1VCG7etTOG

I don't know if there is some change in protocol or something else? I can't take it to Huawei service center as they will tell me phone bluetooth is working fine. Should I try to hard reset? What other options I can try?

philips77, 2018-12-19:
Did you try advertising in nRF Connect on the broken phone?
Do this:
1. On the Huawei 8 in nRF Connect go to Configure GATT Server and select any configuration from the top menu.
2. Make sure that Settings -> Connectivity -> Show incoming connections is enabled (by default it is).
3. Open ADVERTISER tab and create a packet with Local Name, make it Connectable. Start advertising. - check if you won't get any error here
4. On another phone try scanning for that name.
5. On another phone click CONNECT and check if the phones can connect using BLE.
6. On Huawei 8 a new tab should open showing a connection.
This may all work, as only the scanning may be somehow broken. Worth checking if scanning works after this procedure.

And you may try factory reset, yes. We had here Samsung S3 which after factory reset started working again, so this may help. But it also may be a problem with the phone's hardware, hard to know.

junto06, 2018-12-20:
I did one by one and this worked on both devices. However this new device doesn't scan my ble device. It shows newly created Local name on other android 8 device but it doesn't show my ble device. This means issue is not with phone bluetooth? May be ble device isn't compatible with new phone ble module?

philips77, 2018-12-20:
No, it looks like it's something wrong with the firmware part responsible for scanning BLE devices. Or hardware part. No idea. 
The the factory reset then.

closed 2019-01-23 by philips77

clicit, 2019-03-25:
I know it's closed, but any solution here? Same problem with Huawei M5 tablet.
Thanks

philips77, 2019-03-25:
Nope :( Keep in mind that nRF Connect only list Bluetooth LE devices, not ones with Classic Bluetooth.

geeksville, 2019-07-29:
Btw: This problem also occurs with a Pixel 3 XL with Android 9.  A Pixel 2 XL running Android 9 does not show the problem.

singer-yang, 2020-03-26:
> Did you try advertising in nRF Connect on the broken phone?
> Do this:
>
> 1. On the Huawei 8 in nRF Connect go to Configure GATT Server and select any configuration from the top menu.
> 2. Make sure that [...]

Hello all,
I have met the same problem: nRF connect on my cellphone can not find any cellphone devices(only laptops, earphones).
I use this method and my huawei m5pro tablet can find my huawei mate20pro now. It is amazing and I think the key point is to open the gatt server and advertiser scannable. It helps me a lot and thank you again!