nordicsemi/Android-nRF-Connect #92: unexpected Disabling Bluetooth adapter

phatpaul, 2019-02-06:
**Describe the bug**
nRF Connect turns off Bluetooth adapter on my phone unexpectedly.  There is a red bar in the app that shows "Disabling Bluetooth adapter".

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Configure GATT server'
2. Click on 'Sample configuration'
3. Scroll down to 'User Data' (UUID: 0x181C)
4. Edit "First Name" field.
5. Set "Inital Value" text to a ridiculously long string (>500 characters)
6. OK, back, 
7. Go to 'Advertiser' tab and add a connectable advertising packet (I don't think the details matter).
8. Then connect to the USER_DATA (0x181C) service from another device and read the FIRST_NAME characteristic value (0x2A8A)
9.  As soon as I try to read the value, the nRF app shuts down the bluetooth adapter.

**Expected behavior**
I guess there is an error happening, such as exceeded BLE MTU size?  
Shouldn't the GATT server split up the data into multiple packets?

**Screenshots**
[image]

**Versions (please complete the following information):**
 - Android version: 7.1.1
 - Phone model moto x4
 - App Version 4.22.2

**Additional context**

phatpaul, 2019-02-06:
An interesting development: 
If I write to the characteristic from another device, (I'm writing with my custom Cordova app) I'm able to store and read-back 600 characters.  The nRF program does not crash.

I told my Cordova BLE API to write 2495 characters, but when I read it back, I only get the first 600 characters.

It looks like 600 is a significant number.  I've noticed it defined as ESP_GATT_MAX_ATTR_LEN in firmware.

FYI:
I wrote a 2495 character string: 
00005000100001500020000250003000035000400004500050000550006000065000700007500080000850009000095001000010500110001150012000125001300013500140001450015000155001600016500170001750018000185001900019500200002050021000215002200022500230002350024000245002500025500260002650027000275002800028500290002950030000305003100031500320003250033000335003400034500350003550036000365003700037500380003850039000395004000040500410004150042000425004300043500440004450045000455004600046500470004750048000485004900049500500005050051000515005200052500530005350054000545005500055500560005650057000575005800058500590005950060000605006100061500620006250063000635006400064500650006550066000665006700067500680006850069000695007000070500710007150072000725007300073500740007450075000755007600076500770007750078000785007900079500800008050081000815008200082500830008350084000845008500085500860008650087000875008800088500890008950090000905009100091500920009250093000935009400094500950009550096000965009700097500980009850099000995010000100501010010150102001025010300103501040010450105001055010600106501070010750108001085010900109501100011050111001115011200112501130011350114001145011500115501160011650117001175011800118501190011950120001205012100121501220012250123001235012400124501250012550126001265012700127501280012850129001295013000130501310013150132001325013300133501340013450135001355013600136501370013750138001385013900139501400014050141001415014200142501430014350144001445014500145501460014650147001475014800148501490014950150001505015100151501520015250153001535015400154501550015550156001565015700157501580015850159001595016000160501610016150162001625016300163501640016450165001655016600166501670016750168001685016900169501700017050171001715017200172501730017350174001745017500175501760017650177001775017800178501790017950180001805018100181501820018250183001835018400184501850018550186001865018700187501880018850189001895019000190501910019150192001925019300193501940019450195001955019600196501970019750198001985019900199502000020050201002015020200202502030020350204002045020500205502060020650207002075020800208502090020950210002105021100211502120021250213002135021400214502150021550216002165021700217502180021850219002195022000220502210022150222002225022300223502240022450225002255022600226502270022750228002285022900229502300023050231002315023200232502330023350234002345023500235502360023650237002375023800238502390023950240002405024100241502420024250243002435024400244502450024550246002465024700247502480024850249002495

I received a 600 character string: 000050001000015000200002500030000350004000045000500005500060000650007000075000800008500090000950010000105001100011500120001250013000135001400014500150001550016000165001700017500180001850019000195002000020500210002150022000225002300023500240002450025000255002600026500270002750028000285002900029500300003050031000315003200032500330003350034000345003500035500360003650037000375003800038500390003950040000405004100041500420004250043000435004400044500450004550046000465004700047500480004850049000495005000050500510005150052000525005300053500540005450055000555005600056500570005750058000585005900059500600

philips77, 2019-02-07:
Hi,
Could you ask the question on [DevZone](https://devzone.nordicsemi.com/)? Our support team has more experience when it comes to Bluetooth spec and it's more a generic Bluetooth question than related to nRF Connect. From what I know, when you try to read such big value Long Read subprocedure will be used automatically. Both devices should have a buffer long enough to put it together. I guess there is a limit for such message, but I don't know what the spec says or what the phone/your devices has.
Regarding disabling Bluetooth, it's the fact that you try send that big message causes Bluetooth to crash and it restarts itself. nRF Connect just receives a broadcast that Bluetooth has changed state to disabling/enabling. We could find out the max value, but it may differ on different phones, so I guess it is how it is.
Thanks!

closed 2019-02-07 by philips77