nordicsemi/Android-nRF-Connect #93: Clear Log Icon is hidden in some narrow screen

isyq, 2019-03-01:
**Describe the bug**
At the bottom of log panel, the bar contains four icons: copy, save, share and clear.
In portrait mode, the rightmost 'clear' icon is invisible in my phone, with the screen resolution: 3120 x 1440.
In landscape mode, this icon is visible.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a BLE peripheral
2. Right slip to open log panel
3. View the icons at the bottom of log panel
4. The rightmost icon(clear icon) is invisible

**Expected behavior**
All icons should be visible. Or a 'more' icon should be added to display all available icons.

**Screenshots**
Two screenshots at below. One is for landscape which is OK. The other is portrait, not good.

**Versions (please complete the following information):**
 - Android version: [e.g. 9.0.0]
 - Phone model [e.g. Huawei Mate 20 Pro]
 - App Version [e.g. 4.22.3]

[image]
[image]

philips77, 2019-03-01:
Hello,
Thanks for reporting this bug. I have seen it on some of our phones as well. I will fix it in the new release.

philips77, 2020-07-31:
Fixed internally.

closed 2020-07-31 by philips77