nordicsemi/Android-nRF-Connect #96: nRF Connect for Mobile code

weixx, 2019-03-18:
Is there any code that can be referenced for Bluetooth 5.0 broadcast big data?

philips77, 2019-03-18:
As stated in the Readme, the source code of nRF Connect is not available.
For how to advertise using Bluetooth 5 Advertising Extension, check out https://developer.android.com/reference/android/bluetooth/le/BluetoothLeAdvertiser and the new methods and classes added in Android Oreo (API 26).

weixx, 2019-03-19:
Thank you very much for your answer.
I checked a lot of documentation and did a lot of testing, but it still didn't work.
My device supports Bluetooth 5.0. After sending the broadcast, the callback is successful(AdvertisingSetCallback.onAdvertisingSetStarted), but the listener receiving the broadcast will never receive the broadcast I sent, but using nRF Connect for Mobile to send the broadcast, my app's broadcast monitor can receive it. I am very confused. I will attach my broadcast code below. Can you give me some advice?
`
AdvertisingSetParameters.Builder parameters = (new AdvertisingSetParameters.Builder())
                .setInterval(AdvertisingSetParameters.INTERVAL_HIGH)
                .setTxPowerLevel(AdvertisingSetParameters.TX_POWER_MEDIUM)
                .setPrimaryPhy(BluetoothDevice.PHY_LE_CODED)
                .setSecondaryPhy(BluetoothDevice.PHY_LE_CODED)
                .setScannable(true)
                .setInterval(320);

AdvertisingSetCallback callback  = new AdvertisingSetCallback() {
            @Override
            public void onAdvertisingSetStarted(AdvertisingSet advertisingSet, int txPower, int status) {
                advertisingSet.enableAdvertising(true, 300, 255);
            }

            @Override
            public void onAdvertisingSetStopped(AdvertisingSet advertisingSet) {
            }
        };

AdvertiseData data = new AdvertiseData.Builder().addServiceData(ParcelUuid.fromString("0000fff0-0000-1000-8000-00805f9b34fb"),
                "data".getBytes())
                .setIncludeDeviceName(true)
                .build();

BluetoothAdapter.getDefaultAdapter().getBluetoothLeAdvertiser().startAdvertisingSet(parameters.build(), data, null, null, null, callback);
`
Thanks again.

philips77, 2019-03-19:
How about:
```java
[...]
.setInterval(320)
.setLegacyMode(false); // <- calling this
```
Also, in Andvertising Extension, a packet can be either connectable or scannable, or none of those. Legacy packets were either scannable and connectable or not connectable and not scannable.

Wisniowyblazko5, 2019-05-16:
Hell

closed 2019-05-16 by philips77