nordicsemi/Android-nRF-Connect #100: Automated test: XML file include

kruehn, 2019-07-05:
I am writing automated tests with nRF connect. When running a test I need to specify the tests inside an XML file.
When writing multiple tests in multiple files I found no way to include fragments, for example expected service or advertisment data. Thus, I need to copy & paste it in each XML file.

It would be great if the tests would support XML includes. I tried to use XML standard:
```
<!DOCTYPE doc [
<!ENTITY chap1 SYSTEM "CleanEnvironment_import.xml">
]>

``` and then inside the file:
`&chap1;
`
This does not seem to work. Implementing this or another way for imports would be great.

philips77, 2019-07-05:
Hi, 
First of all, thanks for using automated tests and for the feedback!
I see a potential problem with your request. Currently, the batch script is just copying the xml file to Android phone and starts the test. With nested xmls, I would have to allow specifying multiple files in the batch and copying all of them. I think it's doable.
Also, a bigger problem is that I'm using http://simple.sourceforge.net/ for (de)serialization, and this library doesn't seem to support nested entities, which would mean, that I have to implement it on my own.

Although your feature is very reasonable, I'm now buried with another tasks and, as you perhaps have noticed, don't even have time to do bug fixing in nRF Connect :(
But I'll keep this issue open as a reminder.

PS. Any other feedback regarding automated tests? Still setting up the GATT server is unimplemented.

kruehn, 2019-07-05:
Hi,
thank you for quick response, although it is not what I hoped for.
Actually, we are not using a batch script provided by nordic but adb to push the files onto the phone. I tried the include and also pushed multiple files so that was not a problem. 
I see that trade off between effort and value is not so good since the simple framework does not support that.

I don't know how the test runner in nRF connect works. Maybe it would be a solution to specify two files: The actual test file as it is today plus a seconds one containing common data. The test runner could try to find elements in the common data which are not found in the test file. This would kind of emulate a include with the limitation that exactly one file could be included ... hmm. Would be good enough for me but also some effort and limitations in the result.

I guess we will implement a small build step before uploading our test file to nrf and do code replacement on the fly before uploading the file to the phone. But thank you for your answer.

Regards
Tobias

philips77, 2019-07-05:
You may already specify extras in the command (in batch we use -E for that, afair, but you can check for the extra is passed to the service. Not ideal, but at least not in the xml.
I'll come back to you when i resume work on the project.