nordicsemi/Android-nRF-Connect #101: Retrieve nRF Logger output after automated tests

se-abirke, 2019-07-19:
It would be great to be able to retrieve the detailed output of nRF Logger after executing automated tests.

We have had quite some misleading test results where it seems that a certain test step fails even though from the logs it turns out that there has been an error right before that step. (Some of the failures where rather inexplicable, too, as for example error 14 _unknown_ after a sleep instruction). An option to increase the verbosity of the test results or to put an additional log file next to it would be a huge benefit for debugging.

Is there any chance this can be a feature?

Thank you so much

se-abirke, 2019-07-19:
Thank you for your quick reply. I should have been more precise. I am looking for an automated way to do fetch the log files. Similarly to how result files in automated nRF Connect tests are fetched using adb, I would like to fetch the log files using adb. Do you think there's a possibilty to do this?

philips77, 2019-07-20:
Hmm.. on a rooted phone it should be possible. Data are stored in sqlite. But there's no API to export the log from adb yet.

se-abirke, 2019-07-22:
Okay. Unfortunately, we need to test on non-rooted phones right now. I'll see if we can come up with a way of getting the data from sqlite. Thank you for pointing that out.

If an API to extract the data will be a feature some day, it will certainly be very helpful.

philips77, 2019-11-08:
Hi,
I've resumed working on nRF Connect and looked at your issue. The easiest way I can do this is by appending the detailed log to the end of the result file, what do you thing about it?

Here's an example based on very simple XML from #107: 

```
08-11-2019 14:30:15
Version: 4.22.3
Device: Pixel 2, Android version: 10 (QP1A.191005.007.A1)

Starting 'Check connectivity of DE:16:7A:DE:DE:A1' for 'Run-Test'
   Connect...OK
   Disconnect...OK

nRF Connect Test, 2019-11-08
Test Connect (connect.xml)
A	14:30:15.710	Starting 'Check connectivity of DE:16:7A:DE:DE:A1' for 'Run-Test'
A	14:30:15.722	Connect...
V	14:30:15.756	Connecting to DE:16:7A:DE:DE:A1...
D	14:30:15.756	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	14:30:17.208	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	14:30:17.208	Connected to DE:16:7A:DE:DE:A1
A	14:30:17.236	Disconnect...
V	14:30:17.245	Disconnecting...
D	14:30:17.245	gatt.disconnect()
D	14:30:17.257	[Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	14:30:17.257	Disconnected
V	14:30:17.259	Cleaning up...
D	14:30:19.268	gatt.close()
D	14:30:19.284	wait(200)
A	14:30:19.492	'Run-Test' completed
```

kruehn, 2019-11-08:
Not my bug but I would like to mention that attaching the log to the XML output would be really great.

se-abirke, 2019-11-08:
Hi Aleksander,

thank you so much for reconsidering this. Your suggestion is great. We need to investigate the failures manually anyway, so detailed information in the results file totally helps us out.

philips77, 2020-07-31:
Fixed in 4.23, as far as I remember.

closed 2020-07-31 by philips77