nordicsemi/Android-nRF-Connect #102: Add ability to select scan PHY

goldwake, 2019-09-04:
**Is your feature request related to a problem? Please describe.**
When developing and testing devices using coded phy, it is difficult to check which PHY a device is advertising on

**Describe the solution you'd like**
In the Scanner tab, it would be great to be able to select which PHY to scan on or, at a bare minimum see which PHY(s) a device is advertising on 

**Describe alternatives you've considered**
Using an NRF52840 DK and nrf connect on the PC but it doesn't seem to be supported there. Using a NRF52840 DK and scanning only on one PHY at a time then printing devices out over RTT, this is time consuming and annoying switching between PHYs

philips77, 2019-09-05:
Don't you have this info after expanding the device row?

goldwake, 2019-09-05:
I can see that it does but only sometimes. It seems to depend on which advertising message it receives first when the device is advertising on multiple PHYs

philips77, 2019-09-12:
Historical advertising packets from the same device can be viewed in the Device Details screen. The main Scanner tab shows only the most recent advertising packet and refreshes every second, so some packets may be skipped.
But I understand that you may want to see only packets sent on PHY2 or PHY Coded. I'll add it to my backlog. Thanks for the feedback.

philips77, 2022-03-16:
In 4.25.1 you may select Settings -> Scanner -> Physical Layer (PHY) to select only LE Coded. However, this seem not to work correctly on all phones. Some (most?) just use LE 1M as if no setting was set.

The new version added a filter to only show devices advertising with Advertising Extension. Perhaps this may help.

closed 2022-03-16 by philips77