nordicsemi/Android-nRF-Connect #103: want source code for Android-nRF-Connect

majidjamhsidi, 2019-10-03:
hi guy's , why don't have source code for Android-nRF-Connect ?????
if any body have source code please send uploaded notify sent for me

philips77, 2019-10-03:
Hi,
nRF Connect is a tool for developers, not a reference design. We do release source code for all other mobile apps, as they demonstrate how developers may use some features, how may create their own apps, etc. We also provide open source libraries for both Android and iOS. 
May I ask why do you need the source code for nRF Connect? Maybe we can help you in other way.

closed 2019-10-22 by philips77

mnabilurrehman, 2019-11-17:
@philips77 I am developing an android app to receive data from nrf Bluetooth low energy. Can you please guide from where i can find help or any code to do so? As you also mentioned you release source code can you please share the link? Thanks

philips77, 2019-11-18:
Have a look at nRF Blinky or nRF Toolbox on our GitHub account.