nordicsemi/Android-nRF-Connect #104: Zephyr DFU doesn't resume

nwsetec, 2019-10-09:
**Describe the bug**
If a Zephyr DFU fails at some point the DFU can't be resumed without restarting the Zephyr device.

Steps to reproduce the behavior:
1. start a DFU with nRFConnect App Version 4.22.3
2. kill the app once the DFU starts
3. Reopen the app and attempt to start the DFU
4. Note that DFU process fails immediately and doesn't start or resume DFU

**Expected behavior**
DFU process starts or resumes

**Versions (please complete the following information):**
 - nRFConnect App Version 4.22.3
- Zephyr device built with 1.14.0 with mcumgr 

**Additional context**
We have our own Android app that has the DFU library included and it does resume with the BLE Zephyr device used for the above issue.

nwsetec, 2019-10-09:
I also noticed that after 'killing' the app after the DFU started the DFU will complete if left to finish.  It seems there's some process still running even though the app doesn't visually appear to be running.

philips77, 2019-10-09:
Thanks for reporting the issue. I will investigate it. The app hasn't been updated for a while now, as I was busy with other tasks, but I'll resume working on it soon.

philips77, 2020-07-31:
Fixed internally.

closed 2020-07-31 by philips77