nordicsemi/Android-nRF-Connect #107: Command <connect> does not work on Android 10 (timeout)

simonratner, 2019-11-05:
**Describe the bug**
A test that used to run fine on Android 8 no longer works on Android 10 - always fails with "Connect timeout" on the &lt;connect&gt; command.

**To Reproduce**
Steps to reproduce the behavior: 
Run the following script, passing `-E address AA:BB:CC:DD:EE:FF` with a valid address.
```
<test-suite description="Test Connect">
  <set name="address" value="00:00:00:00:00:00" />

  <target id="test_target" address="${address}" />

  <test id="test_connect" description="Check connectivity of ${address}" target="test_target">
    <connect />
    <!-- more commands omitted -->
    <disconnect />
  </test>

  <run-test ref="test_connect" />
</test-suite>
```

**Expected behavior**
The connection succeeds.

**Screenshots**

Failing on Android 10:

[image]

Working on Android 8:

[image]

**Versions (please complete the following information):**
 - App version: 4.22.3
 - Android version: 10.0 (works on 8.0.0)
 - Phone model: Pixel 3

**Additional context**
As you can see in the screenshots, one difference between the `connectGatt` calls seems to be the `opportunistic = true` param which appears on Android 10. Maybe this has something to do with it?

simonratner, 2019-11-05:
Note that on Android 10, even though the test cannot connect to this device, connecting manually from the nRF Connect app works fine.

philips77, 2019-11-05:
Yes, seems like the issue is with the `opportunistic` parameter. It should be set to `false` in Automated Tests and only `true` when another app is trying to connect while nRF Connect is running in background. I will fix it in the upcoming release. 
Thank you for the feedback!

philips77, 2019-11-08:
Fixed internally.

closed 2019-11-08 by philips77