nordicsemi/Android-nRF-Connect #109: Custom 16-bit services cannot be configured in the GATT server

simonratner, 2019-11-21:
**Describe the solution you'd like**
Able to specify a 16-bit UUID for a service in the GATT server, to be advertised.

**Describe alternatives you've considered**
Using 128-bit UUID base value to represent the 16-bit UUID.

philips77, 2020-07-28:
Hi,
I've resume working on nRF Connect. Could you elaborate more on this feature you want?
I just tested and it's possible to create a service/characteristic with 128-bit UUID matching Base Bluetooth UUID in the GATT Server configuration. In Advertiser (you also mentioned that) you can even type just the 2-byte value, or even 4-byte value.
Unknown UUIDs, even if they match Base Bluetooth UUID, are printed in long form as 128-bit.

philips77, 2020-07-30:
I don't know if that's what you wanted, but in the next version it will be possible to create a service/characteristic/descriptor by providing 16- or 32-bit UUID. If not, please reopen. Fixed internally.

closed 2020-07-30 by philips77