nordicsemi/Android-nRF-Connect #111: Could not create log file

kruehn, 2019-12-10:
My Samsung s10e upgraded  to Android 10 today. When I am using nRF connect, then change to the log screen press the save button, I get "Could not create log file."

I tried to reinstall it, but this had no effect. When I press the save button after reinstalling, Android asks for permision. After allowing access to the file system, I can also see that permission in the Android app settings menu. But still I cannot save the log.

The nRF logger app is able to write the log.

I cannot prove it any more, but I think it did work before the Android update.

philips77, 2019-12-11:
Thank you for pointing out. I don't know how u have missed it. The issue will be addressed in the next release. For now, please use Share instead.

kruehn, 2019-12-11:
Is this also related to problems with automated tests? We currently have a test setup which uploads XML filed to nrf, runs a test and gets the results from it. But we always get the message like "Unable to prepare test results" or so.

philips77, 2019-12-11:
Most probably. After the result is complete, the app saves it in the file storage making it available to be pulled to a PC.
This works for me in debug mode, that's why I missed it before.

kruehn, 2019-12-11:
Thank you for your quick response. Do you have an idea, when your next release is coming?

philips77, 2019-12-11:
Regarding Automated Tests, make sure you are using the internal app folder in the script. Have a look here: https://github.com/NordicSemiconductor/Android-nRF-Connect/blob/eaf00a8fddcbf52abb28d5d07880376d5d497885/documentation/Automated%20tests/test.bat#L150
It used to be like that:
https://github.com/NordicSemiconductor/Android-nRF-Connect/blob/d858f4afd08e928e7f4e355eb62e2b8809b32acf/documentation/Automated%20tests/test.bat#L129 and this will no longer work on Android 10.

The other issue, with saving logs and graph, will be fixed today or tomorrow.

kruehn, 2019-12-11:
Ah, ok. Is this backward compatible?

philips77, 2019-12-11:
Yes it is. Should work on android 4.3 as well.

closed 2019-12-11 by kruehn