nordicsemi/Android-nRF-Connect #112: command 'bond' failed due to pairing dialog open on Android

kruehn, 2019-12-11:
When executing automated test with nRF connect, we call the <bond ...> command in certain situations.  Many Android devices then open a pairing dialog which blocks further operation. Is there any workaround?

This can currently be reproduced with all our HTC and XiaoMi (Android 9) phones as well as with Samsung S10e with Android 10.

philips77, 2020-07-31:
No... unfortunately nothing can be done about it. You may lower the security requirement to Just Works, but even then some phones show the dialog to confirm bonding. This is managed by the Android system and apps have no way of overriding it.

Btw, do you have any other feedback regarding Automated Tests?

philips77, 2020-07-31:
And I'm sorry for such late reply. I was buried with other tasks and only now have time to go through these issues.

closed 2020-07-31 by philips77