nordicsemi/Android-nRF-Connect #114: Automated test fails due to lack of permission

LindsayO94, 2020-01-28:
Running automated test using test.bat script - was working several months ago but after either an app or android version update, stopped working. Tried uninstalling and reinstalling the app, no change.

nRF Logger shows:

Test suite parse error (test_script.xml)
E 11:46:51.898 Parsing test suite failed:
E 11:46:51.903 /sdcard/NordicSemiconductor/Test/test_script.xml: open failed: EACCES (Permission denied)
E 11:46:51.909 open failed: EACCES (Permission denied)
E 11:46:51.924 Error while preparing result file
E 11:46:51.931 Permission denied

I removed the space in the folder name in case that was causing issues - didn't help.
The file is in the correct folder, nRF connect just can't access it. I went to the advanced permissions menu in settings and tried to give it storage permissions, but it wasn't in the list of apps I could give permissions to.

 - Android version: 10
 - Phone model Pixel 2
 - App Version 4.24.1

LindsayO94, 2020-01-28:
Backed up to version 4.22.3 and it works fine. Also tried on another phone (android 9) and it did not work with the latest version of the app.

philips77, 2020-01-29:
You need to change paths in the script you run to internal folder of the app. Have a look at this:
https://github.com/NordicSemiconductor/Android-nRF-Connect/blob/eaf00a8fddcbf52abb28d5d07880376d5d497885/documentation/Automated%20tests/test.bat#L150
You're using the NordicSemiconductor folder, which used to work before, but due to changes in the new Android, this won't work now.

philips77, 2020-01-29:
Replace all occurrences of this path in the script.

LindsayO94, 2020-01-29:
Thank you so much! Was there any change to the unbond operation in the latest releases? Doesn't seem to be working now.

philips77, 2020-01-29:
Oh, didn't know. Unbonding was using a hidden API, perhaps it's blacklisted now and can't be used by apps. That must have come one day. Does it work for you on Android 9 and older?

LindsayO94, 2020-01-29:
No, doesn't work on Android 10 - says "device.removeBond() (hidden)" in the log, and the test result indicates a failure during the unbond step.

This isn't too much of an issue since bonding to an already bonded device is allowed, but I'm now also experiencing issues with discover-services as well (fails the test step, but the service list printed in the log looks normal to me?).

Unbond log:
A 10:23:39.775 Starting 'Test' for 'Test Streaming a Recipe'
A 10:23:39.781 Scan for device...
V 10:23:39.790 Starting scan...
D 10:23:39.801 scanner.startScan(null, LOW_LATENCY, callback)
V 10:23:41.326 Stopping scanning...
D 10:23:41.336 scanner.stopScan(callback)
I 10:23:41.345 Device found: C3:77:9F:BE:7A:AC
A 10:23:41.352 Unbond...
V 10:23:41.360 Removing bond information...
D 10:23:41.360 device.removeBond() (hidden)
W 10:23:42.363 Connection timeout
V 10:23:42.380 Cleaning up...
W 10:23:44.395 'Test Streaming a Recipe' failed
I 10:23:44.439 Output file saved (ble_test_result.txt):

29-01-2020 10:23:39
Version: 4.24.1
Device: Pixel 2, Android version: 10 (QQ1A.200105.002)

Starting 'Test' for 'Test Streaming a Recipe'
   Scan for device...
10:23:41.317 C3:77:9F:BE:7A:AC -64 03190000020106080962652E77656C6C11FFFFFFFF03000044E700200000000078FF
   OK
   Unbond...FAIL

After removing the unbond, log with failed discover-services:
A 10:49:21.137 Starting 'Test' for 'Test Streaming a Recipe'
A 10:49:21.153 Scan for device...
V 10:49:21.167 Starting scan...
D 10:49:21.174 scanner.startScan(null, LOW_LATENCY, callback)
V 10:49:22.096 Stopping scanning...
D 10:49:22.106 scanner.stopScan(callback)
I 10:49:22.120 Device found: C3:77:9F:BE:7A:AC
A 10:49:22.126 Bond...
V 10:49:22.142 Create bond request on already bonded device...
I 10:49:22.142 Device bonded
A 10:49:22.144 Connect...
V 10:49:22.163 Connecting to C3:77:9F:BE:7A:AC...
D 10:49:22.163 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D 10:49:24.887 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 10:49:24.887 Connected to C3:77:9F:BE:7A:AC
A 10:49:24.905 Refresh...
V 10:49:24.915 Refreshing device cache...
D 10:49:24.916 gatt.refresh() (hidden)
I 10:49:24.916 Cache refreshed
D 10:49:24.919 wait(1000)
I 10:49:25.719 PHY updated (TX: LE 2M, RX: LE 2M)
A 10:49:25.929 Discover-Services...
V 10:49:25.941 Discovering services...
D 10:49:25.941 gatt.discoverServices()
D 10:49:25.993 [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
D 10:49:26.005 [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDED (12)
I 10:49:26.005 Device bonded
V 10:49:26.007 Cleaning up...
I 10:49:26.530 Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D 10:49:27.421 [Callback] Services discovered with status: 0
I 10:49:27.421 Services discovered
V 10:49:27.470 Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
- Central Address Resolution [R] (0x2AA6)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
[... 40 lines cut ...]
D 10:49:27.474 gatt.setCharacteristicNotification(e3b982d3-8d6f-4ca6-a5ad-79c05e6f6586, true)
D 10:49:27.475 gatt.setCharacteristicNotification(e3b982d4-8d6f-4ca6-a5ad-79c05e6f6586, true)
D 10:49:27.476 gatt.setCharacteristicNotification(f89407da-6b00-413e-9c39-1ba7b312ff37, true)
I 10:49:27.515 Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
V 10:49:28.025 Disconnecting...
D 10:49:28.025 gatt.disconnect()
D 10:49:28.063 [Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I 10:49:28.063 Disconnected
D 10:49:28.063 gatt.close()
D 10:49:28.075 wait(200)
W 10:49:28.286 'Test Streaming a Recipe' failed
I 10:49:28.350 Output file saved (ble_test_result.txt):

29-01-2020 10:49:21
Version: 4.24.1
Device: Pixel 2, Android version: 10 (QQ1A.200105.002)

Starting 'Test' for 'Test Streaming a Recipe'
   Scan for device...
10:49:22.092 C3:77:9F:BE:7A:AC -63 03190000020106080962652E77656C6C11FFFFFFFF03000044E700200000000078FF
   OK
   Bond...OK
   Connect...OK
   Refresh...OK
   Discover-Services...FAIL

LindsayO94, 2020-01-29:
Ah, I think the test was cleaning up too quickly. Adding a sleep at the end fixed this issue.

philips77, 2020-01-29:
Does the service discovery fails on timeout? Try extending it by setting timeout parameter.

LindsayO94, 2020-01-29:
As far as I can tell, no. I extended it to 10 seconds (which is far longer than it takes, looking at the log timestamps) and this issue still happens intermittently. I am curious why the test seems to want to clean up before the results from the service discovery are printed. I'm also curious why it says that the device is already bonded and then seems to go through bonding anyway.

A 10:49:22.126 Bond...
V 10:49:22.142 Create bond request on already bonded device...
**I 10:49:22.142 Device bonded**
A 10:49:22.144 Connect...
V 10:49:22.163 Connecting to C3:77:9F:BE:7A:AC...
D 10:49:22.163 gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D 10:49:24.887 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 10:49:24.887 Connected to C3:77:9F:BE:7A:AC
A 10:49:24.905 Refresh...
V 10:49:24.915 Refreshing device cache...
D 10:49:24.916 gatt.refresh() (hidden)
I 10:49:24.916 Cache refreshed
D 10:49:24.919 wait(1000)
I 10:49:25.719 PHY updated (TX: LE 2M, RX: LE 2M)
A 10:49:25.929 Discover-Services...
V 10:49:25.941 Discovering services...
D 10:49:25.941 gatt.discoverServices()
**D 10:49:25.993 [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
D 10:49:26.005 [Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDED (12)
I 10:49:26.005 Device bonded**
V 10:49:26.007 Cleaning up...

philips77, 2020-01-30:
Ok, I know what's going on. This seems to be an issue in nRF Connect. 
When the bond state changes to BONDED, an internal method `operationCompleted()` is called. At that moment, the service discovery is waiting for operation to finish, so at this moment is resumes. However, services are still `null`, so service discovery thinks it has failed. I have to add a flag indicating that bonding wasn't requested, so operation should not be mark as completed. I will address it in the next release, but I can't promise any date now.

philips77, 2020-01-30:
Hi, could you check if the attached version fixes the issue with service discovery?
[nRF Connect 4.24.1-fixed.apk.zip](https://github.com/NordicSemiconductor/Android-nRF-Connect/files/4133615/nRF.Connect.4.24.1-fixed.apk.zip)
That was a quick fix, but I didn't test it in any way.

LindsayO94, 2020-01-30:
I did get a "GATT ERROR" when trying to connect during one run, but other than that it seems to be working well! May have just been human error, perhaps I reset the device too late and it was in the middle of connecting.

closed 2020-01-31 by philips77