nordicsemi/Android-nRF-Connect #115: Size of Byte Array Defaults to 18 Bytes if the Actual Size is Bigger Than 20 Bytes

TheFrozenDuck, 2020-01-30:
I am using version 4.24.1 of nrfConnect on Android and witnessed an odd issue while transmitting a byte array to my characteristic. When the byte array's length exceeds 20 bytes only the first 18 bytes are transmitted, rather than 20. Furthermore, there is no notice, that the data has been truncated.

philips77, 2020-01-31:
It looks like you are trying to send more than 20 bytes. If MTU is set to 23, the maximum payload size is 20 bytes. 3 bytes are used for handle number (2) and opcode (1).
When you try sending more than MTU-3 bytes using Write Request type (aka Write With Response or WRITE_DEFAULT  on Android), the system is automatically using Long Write Procedure (aka Queued Write). In Long Write procedure payload is split into multiple packets, each maximum 18 bytes (2 bytes are used for offset). First n-1 packets sent are Prepare Write Request followed by Execute Write Request, which contains a flag: execute or cancel. If the device does not support Long Write (a buffer is required to store not executed data), it may send an error and the write request will be cancelled. You actually should not receive those 18 bytes at all in your app, as it has been cancelled, after all, so stack should disregard them.
[image]
Here's a screenshot from a sniffer. The device (uart sample from Nordic SDK 16) rejects Long Write by sending error 0x80. The error code seems invalid, but nevertheless Android cancels write.

Write Command (aka Write Without Response) is only able to send MTU-3 bytes. Long Write is not possible. In nRF Connect, even if you have longer packet, it will be trimmed to fit the package. To switch write type click the arrow next to Advanced on the "Write Value" dialog.

TheFrozenDuck, 2020-02-04:
Thank you for the information. I'll close this ticket.

closed 2020-02-04 by TheFrozenDuck