nordicsemi/Android-nRF-Connect #116: Keyboard Closing When Entering a Value in Landscape Mode

TheFrozenDuck, 2020-01-30:
**Describe the bug**
When trying to enter a value for a transmission to a characteristic the Android keyboard instantly closes.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to a BLE device
2. Click on the "upload" button to open the "Write Value" dialog
3. Turn the display to landscape
4. Try to enter a value
5. The keyboard should now instantly close

**Expected behavior**
The keyboard should not close.

**Screenshots**
n/a

**Versions (please complete the following information):**
 - Android version: 10
 - Phone model: Google Pixel 3XL
 - App Version: 4.24.1

**Additional context**
n/a

philips77, 2020-01-31:
Yeah... That's true. Using the app on a phone in landscape is not very convenient. When the keyboard pops up, the view resizes itself hiding the text filed that has focus causing it to loose focus, which makes the keyboard disappear. I would say it's a minor issue, as you always may rotate the screen to portrait mode. But still an issue.

TheFrozenDuck, 2020-02-04:
What is the proper procedure from now on, shall I keep the ticket open?

philips77, 2020-02-05:
I'd leave the issue open, as the issue is annoying. Otherwise I would forget about it. However, I don't think I will fix this any time soon.