nordicsemi/Android-nRF-Connect #117: Current version of Android-BLE-Library in the app?

robwalkerco, 2020-02-29:
We are currently able to replicate the issue discussed at https://github.com/NordicSemiconductor/Android-BLE-Library/issues/6 in the latest version (4.24.1) of the nRF Connect app.

I wanted to check if the nRF Connect app is using a version of Android-BLE-Library where that issue should be resolved, or whether it is still using a version with the bug?

We have been seeing this issue while using a different Android BLE library, so I wanted to confirm that the nRF Connect app was not susceptible to the issue before attempting to implement the underlaying library myself.

philips77, 2020-03-01:
nRF Connect isn't using BLE library. 
How can we replicate your issue? Are the notifications missing in the log? How fast are you sending them?

robwalkerco, 2020-03-01:
We are connecting to up to two peripherals that send notifications notifications very rapidly. 

When the notifications are inspected, at appears that some notifications are missing and some are duplicated. 
The issue only happens on some devices. Iphone7 and Samsung S7 are effected, while iphone 10 and pixel 2 are not. 

The peripherals are setting the connection interval to 15ms, which I have recently discovered may be the cause of this issue. I've asked the peripheral developers if they can test with a different connection interval (perhaps 30ms). Does that sound like it may be the cause if the issue?

philips77, 2020-03-02:
On Android and iOS there's a problem with fast notifications, as the value received is inside `BluetoothGattCharacteristic` or `peripheral(:didUpdateValueFor:)` respectively. The same field is later reused when a new notification is received. When notifications come too fast the value may get overwritten before it's read, causing 2 handlers to read the same data. There's pretty much nothing that can be done, except from:
1. Values should be read as fast as possible just after receiving.
2. A default `Handler` (on Android) or `DispatchQueue` (on iOS) should be used to read those values. Using another, dedicated handler/queue causes additional delay.
3. Slowing down the notifications (this does not mean increasing connection interval, as notifications can be sent in a single interval, so connection interval does not matter).

That's why in the new BLE library we obtain a reference to the characteristic value in the first line: 
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/e95a814f4c0ecd08a75e5f0e53c68e2b8dbdf70c/ble/src/main/java/no/nordicsemi/android/ble/BleManagerHandler.java#L2025

Both notifications and write without request are not reliable in that sense. If you cannot afford loosing a packet, you need either to implement app-level checks (CRC), or use indications and write with response, which are much slower (only one packet per 2 connection interval vs many packets in a single interval), but will not loose data.

**Note:** On Android it's even worse than that, as the value you send and receive both use the same reference inside `BluetoothGattCharacteristic`, so it may happen that you write an incorrect value it a notification is received after setting the value and before it's actually sent:
```java
characteristic.setValue(value);
// <- notification comes at this moment that changes the value
gatt.writeCharacteristic(characteristic);
```
On iOS the value you send is given as additional parameter, so at least this issue does not appear.

robwalkerco, 2020-03-02:
Thanks @philips77 for your response. Super helpful!

Is the notification handling in the nRF Connect app written in a similar way?

I'm trying to tell if by using Android-nRF-Library we would be improving upon the handling in nRF Connect, or if it would be the same. We are seeing duplicate notification issues in the nRF Connect app, so perhaps the only solution we have is to send the data slower or use indications instead of notifications?

Thanks for your help

philips77, 2020-04-23:
Hi, sorry for not replying before. I tried to base the BLE Library on my experience gained on writing nRF Connect. The app was written before the lib, and is doing lots of things when a notification comes: logging, updating macros, displaying values in UI, sending value to automated tests service, etc, so I guess the value can be replaced by an incoming notification.
Sending data slower should eliminate the issue.

closed 2020-04-23 by philips77