nordicsemi/Android-nRF-Connect #118: RSSI monitor chart (Scanner) X-axis values not visible

wbidus, 2020-03-24:
**Describe the bug**
When using Light theme, X-axis values of the RSSI monitor chart are not visible. They are visible in Dark theme (white font on black background), so i suspect that light mode may be using white font on white background.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Settings
2. Click on Theme
3. Select Light
4. Restart app
5. Open SCANNER tab
6. Swipe left->right to open RSSI monitor chart
7. Look on the chart. Y-axis values (dBm numbers) are visible. X-axis values (elapsed s) are not.
8. Repeat steps 1-4, but in 3 select Dark. Restart app, check chart again. X-axis is labeled fine.

**Expected behavior**
I expect X-axis values to be visible in RSSI monitor chart.

**Screenshots**
Light theme shows problem, Dark theme is fine.
[image]
[image]

**Versions (please complete the following information):**
 - Android version: 10
 - Phone model: Xiaomi Mi9
 - App Version 4.24.1

philips77, 2020-03-24:
Nice one! 
Consider it fixed in the next release. Thanks!

philips77, 2020-07-28:
Fixed internally.

closed 2020-07-28 by philips77