nordicsemi/Android-nRF-Connect #121: Unwanted automatic connection after pressing DISCONNECT

musalehi, 2020-04-01:
**Describe the bug**
the app constantly tries and connects to a previously connected Bluetooth device after "Disconnect" is pressed or the tab was closed by pressing "x" or the Bluetooth device terminated the connection 

**To Reproduce**
Steps to reproduce the behavior:
1. Scan for any Bluetooth device
2. click "CONNECT"
3. wait for the connection to establish and  services to be discovered
3. press "DISCONNECT"
4. See error

**Expected behavior**
After pressing "DISCONNECT" or intentionally closing a connection tab, it is expected that the phone doesn't try establishing a new connection again. 

**Screenshots**
[image]

**Versions (please complete the following information):**
 - Android version: [10]
 - Phone model [Galaxy S10- SM-G973W]
 - App Version [4.24.1]

**Additional context**
Add any other context about the problem here.

philips77, 2020-04-01:
The app doesn't reconnect on its own. It just reacts to events sent by the system. Switch the log level to DEBUG mode to see more information. Android disconnects the physical link 1 second after last client (`BluetoothGatt`) has disconnected. If a notification is received at that time, it may assume the device is still being used. We would need to see the sniffer log to see the reason why it behaves like this. But I imagine, that when you will test a different device, the behavior will be different.

musalehi, 2020-04-01:
Thanks for the quick response. I have attached a log captured in DEBUG from nrf connect. What I see is that the broadcast is also received that the device is disconnected however, after couple of seconds there is another broadcast for connection which is strange! it appears this is a bug in Android for my S10 device? As for the notification I am not sure if that is the case here. Since this autoconnect behavior is observed even after disconnection is triggered from the node and I am not anticipating the node sending notification after disconnection. I would appreciate your comment on this.

Regards,Mustafa

 
 

  
nRF Connect, 2020-04-01
Empty Example (00:0D:6F:40:D2:6A)
V	12:23:23.282	Disconnecting...
D	12:23:23.282	gatt.disconnect()
D	12:23:23.295	[Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	12:23:23.295	Disconnected
D	12:23:24.320	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	12:23:29.695	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	12:23:29.695	gatt.close()
D	12:23:29.707	wait(200)
V	12:23:29.911	Connecting to 00:0D:6F:40:D2:6A...
D	12:23:29.911	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	12:23:29.929	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	12:23:29.929	Connected to 00:0D:6F:40:D2:6A
V	12:23:29.949	Discovering services...
D	12:23:29.949	gatt.discoverServices()
I	12:23:30.514	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	12:23:30.664	[Callback] Services discovered with status: 0
I	12:23:30.665	Services discovered
V	12:23:30.681	Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
- Database Hash [R] (0x2B2A)
- Client Supported Features [R W] (0x2B29)
Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- System ID [R] (0x2A23)
Bluegiga OTA (1d14d6ee-fd63-4fa1-bfa4-8f47b42119f0)
- ota_control [W] (f7bf3564-fb6d-4e53-88a4-5e37e0326063)
D	12:23:30.681	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
I	12:23:30.730	Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
V	12:23:39.685	Disconnecting...
D	12:23:39.685	gatt.disconnect()
D	12:23:39.701	[Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	12:23:39.701	Disconnected
D	12:23:40.748	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	12:23:43.299	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	12:23:43.307	gatt.close()
D	12:23:43.314	wait(200)
V	12:23:43.518	Connecting to 00:0D:6F:40:D2:6A...
D	12:23:43.518	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	12:23:43.540	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	12:23:43.540	Connected to 00:0D:6F:40:D2:6A
V	12:23:43.553	Discovering services...
D	12:23:43.553	gatt.discoverServices()
I	12:23:44.128	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	12:23:44.274	[Callback] Services discovered with status: 0
I	12:23:44.274	Services discovered
V	12:23:44.290	Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
- Database Hash [R] (0x2B2A)
- Client Supported Features [R W] (0x2B29)
Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- System ID [R] (0x2A23)
Bluegiga OTA (1d14d6ee-fd63-4fa1-bfa4-8f47b42119f0)
- ota_control [W] (f7bf3564-fb6d-4e53-88a4-5e37e0326063)
D	12:23:44.290	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
I	12:23:44.359	Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
V	12:23:48.095	Disconnecting...
D	12:23:48.095	gatt.disconnect()
D	12:23:48.114	[Callback] Connection state changed with status: 0 and new state: DISCONNECTED (0)
I	12:23:48.114	Disconnected
D	12:23:49.187	[Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
D	12:23:50.035	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	12:23:50.035	gatt.close()
D	12:23:50.046	wait(200)
V	12:23:50.249	Connecting to 00:0D:6F:40:D2:6A...
D	12:23:50.249	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	12:23:50.277	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	12:23:50.277	Connected to 00:0D:6F:40:D2:6A
V	12:23:50.288	Discovering services...
D	12:23:50.288	gatt.discoverServices()
I	12:23:50.855	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	12:23:51.115	[Callback] Services discovered with status: 0
I	12:23:51.115	Services discovered
V	12:23:51.135	Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
- Database Hash [R] (0x2B2A)
- Client Supported Features [R W] (0x2B29)
Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- System ID [R] (0x2A23)
Bluegiga OTA (1d14d6ee-fd63-4fa1-bfa4-8f47b42119f0)
- ota_control [W] (f7bf3564-fb6d-4e53-88a4-5e37e0326063)
D	12:23:51.135	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
I	12:23:51.247	Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)

philips77, 2020-07-30:
Hi, sorry for the late reply. As I wrote, nRF Connect just acts and displays state based on the events it get from the system. 
In your case I can say, that the device you connect to does not disconnect from your phone, so the phone assumes it is still connected, or connects again. You may try disabling "Show incoming connections" in Settings -> Connectivity, or disabling GATT server if you have it enabled, but that may only "hide" the effect. The only way to truly disconnect from a device is by asking it to disconnect from the peripheral side, e.g. by creating a special command, that the phone can send, or by pressing reset button. Without this feature it is only the local client (`BluetoothGatt`) that disconnects, but the physical link can still be used by another client in your app, or any other app. nRF Connect opens only 1 client to each device unless DFU or McuManager are user.

philips77, 2020-07-30:
Closing, as this isn't really a bug in nRF Connect.

closed 2020-07-30 by philips77