nordicsemi/Android-nRF-Connect #123: Data issue

jkhax0r, 2020-04-08:
nRF Connect gets very slow to the point of being unusable after a few days of usage.  The cause seems to be that DATA folder via "Settings->App->NRF Connect" shows 5GB+ of data.  When I clear this it seems things are back to normal for a while.

I frequently use nRFConnect at the same time as our app, which is regularly streaming BLE notification data fairly fast.  I suspect that an internal log file is getting huge?  If so could you perhaps add a feature to clear this or simply turn it off?

philips77, 2020-04-09:
There's a button in the log pane to delete the log. In some phones it may be hidden, unfortunately, add they are too narrow (it's a known bug), but you may rotate the phone to landscape and you'll find it in the bottom bar.

philips77, 2020-04-09:
Indeed, the app logs a lot. There's no option to disable logging, as the app was mainly design for debugging, and log is helpful for that.

jkhax0r, 2020-05-29:
It might be that I found the setting to prompt before saving log sessions, but it seems better now.  

Mostly my phone just seems to suck but its often much worse because I frequently have my phone streaming ~3KB/s of notifications at the same time I want to use nRF connect.  Might also be the 100's of BLE device's sitting around in the vicinity and surely polluting my phone's stack.  

Turning off logging completely would be a nice enhancement but can't complain.

The trash icon on the logs was in plain sight on my phone btw.

closed 2020-05-29 by jkhax0r