nordicsemi/Android-nRF-Connect #125: Not possible to import user_definitions.json in 4.24.1

gte1, 2020-04-22:
Using 4.24.1, when trying to import user_definitions.json, any such file is greyed out and not possible to select.
If downgrading to 4.23.0, importing works fine

philips77, 2020-04-23:
Thank you very much for the bug report. I'll address it in the next release. Great that you found a workaround.

philips77, 2020-07-28:
Hi, may I ask what phone are you using and what app? I've just tried on Pixel 4 and Samsung S6 and the native file chooser or Drive and I was able to select json file.

gte1, 2020-07-28:
Huawei P20 Pro, and Android nRFConnect 4.24.1 (downgrading to 4.23.0 made importing possible again)

gte1, 2020-07-28:
Sorry... works fine on my phone... will find the phone that failed

philips77, 2020-07-29:
The issue is in the File Browser app, which does not understand *application/json* Mime Type. A different app can be used to select this file.

closed 2020-07-29 by philips77