nordicsemi/Android-nRF-Connect #128: Is there a source code for Android apps?

hjh2881, 2020-05-29:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

philips77, 2020-05-29:
No, the app is not open sourced.
It's written on Readme: https://github.com/NordicSemiconductor/Android-nRF-Connect#application
Duplicate of #1, #49, #53, #96, #103.

philips77, 2020-05-29:
We have open sourced our other [Android](https://github.com/NordicSemiconductor?q=android&type=&language=) and [iOS](https://github.com/NordicSemiconductor?q=IOS&type=&language=) apps.

closed 2020-05-29 by philips77

hjh2881, 2020-06-01:
Thank you very much.