nordicsemi/Android-nRF-Connect #130: v4.24.1 App crahes every single time when doing DFU on validating stage.

MuzammilPatel, 2020-07-13:
Description-
While doing a DFU of an image on v4.24.1, App crashes at the validation of the image. It is consistent with Android device. I have tested with - OPPO A73 and Samsung Galaxy S7. 
I had v4.22.3 previously and it worked all of the time. 
I also degraded the app from v4.24.1 to v4.22.3 and DFU works but never works on v4.24.1
I am using MCUboot as my bootloader and Zephyr Project as RTOS to build my firmware.

Android Version-
- OPPO A73, Model CPH1725, Android V7.1.1
- Samsung Galaxy S7, Model SM-G930F, Android V7.0

MuzammilPatel, 2020-07-15:
While testing DFU with LG V30+, Android V9, and nRF-app v4.24.1, DFU was working completely fine.

MuzammilPatel, 2020-07-15:
It seems like it is some were related to the Android version.

philips77, 2020-07-15:
Could you try updating using this app: https://play.google.com/apps/internaltest/4699465122035899213
You may find source code on our GitHub, and get some logs that would help. This app is using the latest mcumgr library.

MuzammilPatel, 2020-07-15:
Hi Philips, 
Thanks for responding. I can't access the pay store as my account is not yet been invited.

Link leads to -
     App not available for this account
     Your account hasn't yet been invited to participate in this app's internal testing program.

philips77, 2020-07-15:
If you have Android Studio clone https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager and try this version.

lukasbrchl, 2020-07-16:
Hi, we have just discovered the same issue in our app and device. The v4.24.1 app crashes after validating the image on Android 9, Huawei Mate 9.  The v4.22.3 works without problems.

philips77, 2020-07-29:
Fixed internally.

closed 2020-07-29 by philips77

lukasbrchl, 2020-07-29:
Thanks, Aleksandr, could you please send us APK or should we wait until it appears on Google Play?

MuzammilPatel, 2020-07-29:
Hi, Could I know what caused the issue in the new app when it was not in the previously released versions?

philips77, 2020-07-29:
I suspect 2 reasons. First is proguard rules and lack of a class in the final APK. Second is a bug is the last version of McuManager library. Now I updated to 0.11.0 and Echo, DFU and other features seem to work fine.
I planned to release the new version soon, but will be happy if you test it before. Find the APK attached (you need to unpack it, APKs are not allowed on GitHub).
[nRFConnect 4.24.2-beta.apk.zip](https://github.com/NordicSemiconductor/Android-nRF-Connect/files/4994697/nRFConnect.4.24.2-beta.apk.zip)

lukasbrchl, 2020-07-29:
I just tested the beta app on in our solution and we were able to update the device.