nordicsemi/Android-nRF-Connect #131: Partial characteristic value assertion in macros

reinhardknoll, 2020-07-22:
**Is your feature request related to a problem? Please describe.**
Currently the assert-value macro command has to match the entire characteristic value. This can be problematic if it also contains a non-deterministic field, e.g. time.

**Describe the solution you'd like**
Some form of partial value assertion (e.g. regex, wildcard character or mask) could be useful if you only want to match specific characters in the characteristic value.

**Describe alternatives you've considered**
I'm not aware of any alternative if you don't have control over the characteristic  contents.

**Additional context**
Add any other context or screenshots about the feature request here.

philips77, 2020-07-27:
That is true with some buts:
1. In macros, in contrary to automated tests, an assert does not stop macro, just displays a warning icon (unless "sleep-if/sleep-until" are used).
2. It would be very difficult to make such macro "recordable". It would have to be modified in XML and imported again.

reinhardknoll, 2020-07-27:
1. It could still be useful to have that check if you need it. It's better than not having any assert at all.
2. Does it have to be recordable though? We edit our macros manually anyway.

philips77, 2020-07-30:
Implemented internally. You'll be able to provide any Java-compatible regex in both "value" and "valueString" fields of AssertValue.

E.g.
```xml
<wait-for-notification description="Wait for notification to SMP Characteristic" characteristic-uuid="da2e7828-fbce-4e01-ae9e-261174997c48" service-uuid="8d53dc1d-1db7-4cd3-868b-8a527460aa84">
      <assert-value description="Assert value matches regex" value="0300[0-9A-F]{4}00000000BF61726[0-9A-F]48656C6C6F20[A-F0-9]*FF"/>
   </wait-for-notification>
```

philips77, 2020-07-30:
Regex matching has also been added to Automated Tests to AssertValue tag to "value" attribute. It was there for "valueString" already.

closed 2020-07-30 by philips77