nordicsemi/Android-nRF-Connect #133: [DFU] slowdown with nRFConnect 4.24.3 compared to 4.24.1

marcoster, 2020-08-25:
**Describe the bug**
Since the updates of 4.24.3 (and I guess also 4.24.2) the transmission data-rate during DFU decreased by approximately 50% (~5kB/s instead of ~ 10kB/s on 4.24.1; file size of ~150kB).
It seems like the source are delays of 400msec after each object created (38 objects).

Time with 4.24.1: 15368ms
Time with 4.24.3: 30805ms
38 (objects) * 400ms = 15200ms ~ difference

The secure-DFU from SDK 15.3 is used on the device-side.

**To Reproduce**
Compare Upgrades with the 2 app-versions. On log-level DEBUG the `[DFU] wait(400)` lines occur in 4.24.3 after every created object (also see attached logs).

**Expected behavior**
No delay after creating or some polling mechanism, if possible.
Or, if this is needed by some devices, an configuration option for this delay.

**Versions (please complete the following information):**
 - Android version: 10
 - Phone model: Xiaomi Mi 9
 - App Version: 4.24.3 (maybe 4.24.2)
Also seen on Sony Xperia XZ1 with Android 9 (no comparision made now)

**Log 4.24.1**
```
V	11:57:28.110	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:57:28.110	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:57:28.119	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-02-00-10-00-00
I	11:57:28.165	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
A	11:57:28.166	[DFU] Data object (1/38) created
D	11:57:28.166	[DFU] wait(400)
A	11:57:28.567	[DFU] Uploading firmware...
V	11:57:28.567	[DFU] Sending firmware to characteristic 8ec90002-f315-4f60-9fb8-838830daea50...
V	11:57:28.641	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:57:28.641	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:57:28.842	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-03-01-00-10-00-00-00-E0-51-59
I	11:57:28.842	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-03-01-00-10-00-00-00-E0-51-59
A	11:57:28.844	[DFU] Checksum received (Offset = 4096, CRC = 5951E000)
V	11:57:28.844	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:57:28.844	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:57:28.873	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-01
I	11:57:28.873	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-01
A	11:57:28.873	[DFU] Data object executed
V	11:57:28.873	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:57:28.873	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:57:28.899	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-02-00-10-00-00
I	11:57:28.929	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
A	11:57:28.929	[DFU] Data object (2/38) created
A	11:57:28.929	[DFU] Uploading firmware...
V	11:57:28.930	[DFU] Sending firmware to characteristic 8ec90002-f315-4f60-9fb8-838830daea50...
V	11:57:28.968	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:57:28.968	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:57:29.201	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-03-01-00-20-00-00-09-D1-29-3A
I	11:57:29.202	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-03-01-00-20-00-00-09-D1-29-3A
A	11:57:29.202	[DFU] Checksum received (Offset = 8192, CRC = 3A29D109)
V	11:57:29.202	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:57:29.202	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:57:29.232	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-01
I	11:57:29.232	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-01
A	11:57:29.232	[DFU] Data object executed
V	11:57:29.233	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:57:29.233	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:57:29.261	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-02-00-10-00-00
I	11:57:29.322	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
A	11:57:29.322	[DFU] Data object (3/38) created
A	11:57:29.322	[DFU] Uploading firmware...
```

**Log 4.24.1**
```
A	11:59:04.331	[DFU] Data object info received (Max size = 4096, Offset = 0, CRC = 00000000)
V	11:59:04.358	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:59:04.358	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:59:04.358	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-02-00-10-00-00
I	11:59:04.424	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
A	11:59:04.424	[DFU] Data object (1/38) created
D	11:59:04.424	[DFU] wait(400)
A	11:59:04.824	[DFU] Uploading firmware...
V	11:59:04.824	[DFU] Sending firmware to characteristic 8ec90002-f315-4f60-9fb8-838830daea50...
V	11:59:04.873	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:59:04.873	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:59:05.097	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-03-01-00-10-00-00-00-E0-51-59
I	11:59:05.097	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-03-01-00-10-00-00-00-E0-51-59
A	11:59:05.099	[DFU] Checksum received (Offset = 4096, CRC = 5951E000)
V	11:59:05.099	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:59:05.099	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:59:05.135	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-01
I	11:59:05.136	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-01
A	11:59:05.136	[DFU] Data object executed
V	11:59:05.136	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:59:05.136	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:59:05.155	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-02-00-10-00-00
I	11:59:05.214	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
A	11:59:05.214	[DFU] Data object (2/38) created
D	11:59:05.214	[DFU] wait(400)
A	11:59:05.617	[DFU] Uploading firmware...
V	11:59:05.617	[DFU] Sending firmware to characteristic 8ec90002-f315-4f60-9fb8-838830daea50...
V	11:59:05.688	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:59:05.688	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:59:05.893	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-03-01-00-20-00-00-09-D1-29-3A
I	11:59:05.893	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-03-01-00-20-00-00-09-D1-29-3A
A	11:59:05.893	[DFU] Checksum received (Offset = 8192, CRC = 3A29D109)
V	11:59:05.893	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:59:05.893	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:59:05.921	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-01
I	11:59:05.921	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-04-01
A	11:59:05.921	[DFU] Data object executed
V	11:59:05.921	[DFU] Writing to characteristic 8ec90001-f315-4f60-9fb8-838830daea50
D	11:59:05.923	[DFU] gatt.writeCharacteristic(8ec90001-f315-4f60-9fb8-838830daea50)
I	11:59:05.949	[DFU] Data written to 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 01-02-00-10-00-00
I	11:59:06.013	[DFU] Notification received from 8ec90001-f315-4f60-9fb8-838830daea50, value (0x): 60-01-01
A	11:59:06.013	[DFU] Data object (3/38) created
D	11:59:06.013	[DFU] wait(400)
A	11:59:06.418	[DFU] Uploading firmware...
```

philips77, 2022-03-16:
The long delay was introduced in [DFU Library 1.10](https://github.com/NordicSemiconductor/Android-DFU-Library/releases/tag/v1.10.0) and it improves stability of DFU on some devices.

philips77, 2022-03-16:
3rd party apps, which are using DFU library like Nrf Connect does, can tailor the delay to their needs. nRF Connect is using the default value, that we found working with some of our samples.

closed 2022-03-16 by philips77