nordicsemi/Android-nRF-Connect #134: Oximetry Characteristic data not being parsed to string.

ubdussamad, 2020-09-04:
**Describe the bug**
It's always the case that if a SIG  defined characteristic is used by the server, the data that is sent to the nRF client is **parsed** to a human readable string format from the simple bytes format which is sent under the hood.

For example:
If a server sends a hex array: `00-00-ff-00-ff` 
The app will parse it as:  _**Heart rate: 255 , Something: 255**_

But, in case of the _**PLX Continuous Measurement Characteristic**_  this does not happens, it just displays the hex values it received from the server instead of parsing it.

Even, in case that I am not sending valid data format (SFLOAT), it should (atleast) still display: _**Incorrect Format**_ but it just displays the values.

Is this intentional? or just a mishap? or I totally don't know what I am doing?

**To Reproduce**
Steps to reproduce the behavior:
1. Create a BLE Pulse Oximeter Service (UUID:1822) (I use ESP32 as a BLE server.)
2. Create a **PLX Continuous Measurement Characteristic** (UUID:2A5F) characteristic.
3. Connect to the server and read the characteristic data.
4. Notice that there is NO parsing of the received value.

**Expected behavior**
I expected the value received to be parsed and shown as a string. 
Like:  Oxygen Concentration: 90% , Heart rate: 80BPM
But instead, I just get a hex string displayed as the value. 

**Screenshots**
Click [here](https://imgur.com/a/fGf4Jlx) for the screen shot. (Imgur link)

**Versions (please complete the following information):**
 - Android version: Android 10 (Samsung One UI 2.0)
 - Phone model:  Samsung Galaxy A50
 - App Version: 4.24.3

**Additional context**
- This issue is also pointed out on [StackOverflow](https://stackoverflow.com/questions/55998274/can-pulse-oximeter-service-0x1822-be-shown-on-nrf-connect-for-mobile).
- I am using a ESP32 as a BLE server.

philips77, 2020-09-05:
Hello,
Unfortunately not all Bluetooth sig character have parsers implemented. I'll try to add one you this one in the next release.

ubdussamad, 2020-09-05:
@philips77  Thanks you for your effort, can I in any way help with this? I would love to learn how you guys do these things.

philips77, 2022-08-11:
Hi, perhaps it's a bit late, but I've started working on PLX parsers. They should be available in the next release, planned for tomorrow (Friday release!).

philips77, 2022-08-12:
Fixed internally, will be released as 4.25.4 today.

closed 2022-08-12 by philips77