nordicsemi/Android-nRF-Connect #136: Android 11 not working with automated tests

LindsayO94, 2020-10-07:
**Describe the bug**
When I attempt to run an automated test, I now get an error after updating to Android 11. When I checked the /sdcard/Android/data folder using the Files app, it's empty.

**To Reproduce**
Steps to reproduce the behavior:
1. Have a phone with Android 11
2. Attempt to run a test ("test.bat <test_file_name>.xml")
3. Test does not run

**Expected behavior**
My test should run

**Screenshots**
The output I get is as follows:
=================================================
Nordic Semiconductor Automated Tests batch script
=================================================
Removing old result file...NOT FOUND
Copying "test_duration_seconds.xml" to /sdcard/Android/data/no.nordicsemi.android.mcp/files/Test...FAIL
Error: Device not found.

**Versions (please complete the following information):**
 - Android version: 11
 - Phone model: Pixel 2
 - App Version: 4.24.3

philips77, 2020-10-07:
Thank you for the report. I'll look into that, but as I'm out of office for 3 months, I can do this in January as shortest. So you see your device when you type `adb devices`?

LindsayO94, 2020-10-07:
I do see my device with adb devices. If I try to run the test without it attached, I instead get: "Error: No device connected."

philips77, 2020-10-07:
I guess it's related to that: https://developer.android.com/about/versions/11/privacy/storage
As I said, I'll be in the office next year... Sorry for the inconveniences! You may look for another directory that those files are now located. The app is targeting Android 10.

philips77, 2022-08-12:
We're addressing this issue in the new version of nRF Connect 4.25.4. I've tested it on Android 11, 12 and 13 and seems to be working.
No storage permission needed for the app, as the batch script copies the files to app's local storage on the external storage (`sdcard/data/no.nordicsemi.android.mcp/files/Test/`).
Hope it will work for you!

closed 2022-08-12 by philips77