nordicsemi/Android-nRF-Connect #138: Can't connect while in airplane mode (and any subsequent attempts)

QuentinFarizon, 2020-11-15:
**Describe the bug**
While in airplane mode, connecting to a NRF device fails

**To Reproduce**
Steps to reproduce the behavior:
1. On the phone, switch to airplane mode
2. Activate Bluetooth and location (and even Wifi)
3. Launch scan, scan succeeds
4. Connect to the selected NRF device
5. Connection never succeeds, after 30 seconds an Error 133 GATT error appears

After that, even we quitting airplane mode, connection never succeed anymore.
Restarting fixes the issue most of the time, but not always.

**Expected behavior**
Connection shoudl succeed

**Versions (please complete the following information):**
 Reproduced on :
- a Huawei 20 pro CLT-L09, Android 10
- a Samsung S10, Android 10
App Version : 4.24.3

philips77, 2020-11-15:
Thank you for the report. I'll look into it, but I think it's more likely to be the Android or phone issue...

philips77, 2022-03-16:
I tested it on Pixel 5, and it seems to work, which indicates a phone issue. Sorry for the super long delay. Closing the issue.