nordicsemi/Android-nRF-Connect #141: About

BIGMONK, 2020-12-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Versions (please complete the following information):**
 - Android version: [e.g. 8.0.0]
 - Phone model [e.g. Pixel 2]
 - App Version [e.g. 4.20.0]

**Additional context**
Add any other context about the problem here.

roshanrajaratnam, 2020-12-09:
HI @BIGMONK  if you enter the company identifier which is a 16-bit or 2-octet value and some data it should enable the ok button.

BIGMONK, 2020-12-09:
> HI @BIGMONK if you enter the company identifier which is a 16-bit or 2-octet value and some data it should enable the ok button. Company identifier is a Bluetooth SIG Defined value, for example identifier for Nordic Semiconductor is 0x0059.

Thank you very much

closed 2021-01-19 by philips77