nordicsemi/Android-nRF-Connect #142: 4.24.3 DFU process doesn't start after choosing distribution package

markoaamunkajo, 2021-01-30:
**Describe the bug**
User is trying to upload distribution zip package to update a RuuviTag device.
Firstly app cannot use the My Files built-in file manager, it only shows me content of my music player.
User then installed File Commander from Play Store and is able to select the zip package to open.
But opening file doesn't start the DFU process, it doesn't do anything.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to DFU, choose distrubution package zip file
2. Browse to dfu zip file and select it
3. User goes back to the app but DFU doesn't start.

**Expected behavior**

DFU process should start.

**Versions (please complete the following information):**
 - Android version: 6.0.1
 - Phone model: Samsung S5
 - App Version 4.24.3

philips77, 2021-02-02:
I believe that's because you're exporting from File Commander using *file://* instead of *content://*. The first one is not supported anymore and was deprecated in Android. 
Also, the native My Files should work. Perhaps you may navigate to other folder, or something like that. If that doesn't help, try a different app from Google Play Store.

philips77, 2022-03-16:
Closing due to inactivity.

closed 2022-03-16 by philips77