nordicsemi/Android-nRF-Connect #144: [DFU] New firmware is not confirmed

Qpczyk, 2021-02-10:
**Describe the bug**
After firmware is flashed successfully device boots up with new firmware. Next, after reset, device reverts to the previously flashed firmware because new image was not confirmed.   

Confirming image after boot manually using mcumgr api function in device sw solves the problem.
Problem do not occur after device boots to the new firmware and another dfu is performed, this time with 'confirm only' option selected.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to the device and select dfu option
2. Select option 'Test and Confirm'
3. Update firmware in the device
4. Device reboots with new firmware
5. Reset/power cycle the deevice
6. Device starts firmware revert process and boots with old firmware

**Expected behavior**
After performing dfu with option 'Test and confirm' device boots to the new firmware and firmware is confirmed. Each next reset/power cycle do not lead to firmware revert.

**Screenshots**
--none--

**Versions (please complete the following information):**
 - Android version: 10
 - Phone model Honor 20
 - App Version: 4.24.3

Tested with: nrf52840 Zephyr v2.4.0

**Additional context**
Problem was detected with devices firmware version which was confirmed to be working properly with dfu.

philips77, 2021-02-11:
Hello,
After 4. the app should send Confirm message to the new firmware.
Could you post the logs here?

Also, could you try [nRF Connect Device Manager](https://play.google.com/store/apps/details?id=no.nordicsemi.android.nrfconnectdevicemanager)?

Qpczyk, 2021-02-11:
```
nRF Connect, 2021-02-11
DEV (00:00:00:00:00:00)
V	11:23:23.967	Connecting to 00:00:00:00:00:00...
D	11:23:23.967	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	11:23:28.622	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	11:23:28.622	Connected to 00:00:00:00:00:00
V	11:23:28.644	Discovering services...
D	11:23:28.644	gatt.discoverServices()
D	11:23:28.648	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I	11:23:28.734	PHY updated (TX: LE 2M, RX: LE 2M)
I	11:23:29.066	Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 5000ms)
I	11:23:29.361	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	11:23:29.508	[Callback] Services discovered with status: 0
I	11:23:29.508	Services discovered
V	11:23:29.534	Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
- Client Supported Features [R W] (0x2B29)
- Database Hash [R] (0x2B2A)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Device Information (0x180A)
- Model Number String [R] (0x2A24)
- Manufacturer Name String [R] (0x2A29)
- Firmware Revision String [R] (0x2A26)
SMP Service (8d53dc1d-1db7-4cd3-868b-8a527460aa84)
- SMP Characteristic [N WNR] (da2e7828-fbce-4e01-ae9e-261174997c48)
   Client Characteristic Configuration (0x2902)
D	11:23:29.535	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
I	11:23:29.599	Connection parameters updated (interval: 30.0ms, latency: 0, timeout: 5000ms)
I	11:23:33.703	Connection parameters updated (interval: 50.0ms, latency: 0, timeout: 500ms)
V	11:23:39.624	[McuMgr] Connecting...
D	11:23:39.626	[McuMgr] gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
D	11:23:39.638	[McuMgr] [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
I	11:23:39.643	[McuMgr] Connected to 00:00:00:00:00:00
D	11:23:39.647	[McuMgr] wait(300)
V	11:23:39.951	[McuMgr] Discovering services...
D	11:23:39.958	[McuMgr] gatt.discoverServices()
I	11:23:39.970	[McuMgr] Services discovered
V	11:23:39.977	[McuMgr] Primary service found
V	11:23:39.984	[McuMgr] Requesting new MTU...
D	11:23:39.992	[McuMgr] gatt.requestMtu(515)
I	11:23:40.055	[McuMgr] MTU changed to: 252
D	11:23:40.060	[McuMgr] gatt.setCharacteristicNotification(da2e7828-fbce-4e01-ae9e-261174997c48, true)
V	11:23:40.066	[McuMgr] Enabling notifications for da2e7828-fbce-4e01-ae9e-261174997c48
D	11:23:40.068	[McuMgr] gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
I	11:23:40.155	[McuMgr] Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
I	11:23:40.160	[McuMgr] Notifications enabled
V	11:23:40.166	[McuMgr] Writing characteristic da2e7828-fbce-4e01-ae9e-261174997c48 (WRITE COMMAND)
D	11:23:40.167	[McuMgr] gatt.writeCharacteristic(da2e7828-fbce-4e01-ae9e-261174997c48)
I	11:23:40.176	[McuMgr] Data written to da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 00-00-00-02-00-01-00-00-BF-FF
A	11:23:40.186	[McuMgr] "Operation: READ
Flags: 0
Length: 2
Group Id: 1 (IMAGE)
Sequence Num: 0
Command Id: 0 (STATE)
Message: {}" sent
I	11:23:40.259	[McuMgr] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 01-00-00-86-00-01-00-00-BF-66-69-6D-61-67-65-73-9F-BF-64-73-6C-6F-74-00-67-76-65-72-73-69-6F-6E-65-30-2E-30-2E-30-64-68-61-73-68-58-20-23-BE-E2-59-7D-AE-C9-A2-B3-90-AB-B6-FC-12-CC-A3-20-6E-84-B1-5C-C9-6D-D7-DF-64-29-D5-9A-45-61-D4-68-62-6F-6F-74-61-62-6C-65-F5-67-70-65-6E-64-69-6E-67-F4-69-63-6F-6E-66-69-72-6D-65-64-F5-66-61-63-74-69-76-65-F5-69-70-65-72-6D-61-6E-65-6E-74-F4-FF-FF-6B-73-70-6C-69-74-53-74-61-74-75-73-00-FF
A	11:23:40.292	[McuMgr] "{"images":[{"slot":0,"version":"0.0.0","hash":"I77iWX2uyaKzkKu2/BLMoyBuhLFcyW3X32Qp1ZpFYdQ=","bootable":true,"pending":false,"confirmed":true,"active":true,"permanent":false}],"splitStatus":0}" received
V	11:23:40.300	[McuMgr] Uploading firmware...
V	11:28:19.463	[McuMgr] Writing characteristic da2e7828-fbce-4e01-ae9e-261174997c48 (WRITE COMMAND)
D	11:28:19.469	[McuMgr] gatt.writeCharacteristic(da2e7828-fbce-4e01-ae9e-261174997c48)
I	11:28:19.473	[McuMgr] Data written to da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 02-00-00-32-00-01-00-00-BF-67-63-6F-6E-66-69-72-6D-F4-64-68-61-73-68-58-20-32-62-CB-80-27-12-10-BD-79-44-AB-89-B0-12-CC-96-05-85-8B-05-DD-B3-96-47-78-5C-0D-10-13-9A-A6-5B-FF
A	11:28:19.483	[McuMgr] "Operation: WRITE
Flags: 0
Length: 50
[... 43 lines cut ...]
I	11:29:05.555	Connection parameters updated (interval: 50.0ms, latency: 0, timeout: 5000ms)
I	11:29:06.050	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	11:29:06.211	[Callback] Services discovered with status: 0
I	11:29:06.211	Services discovered
V	11:29:06.233	Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
- Client Supported Features [R W] (0x2B29)
- Database Hash [R] (0x2B2A)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Device Information (0x180A)
- Model Number String [R] (0x2A24)
- Manufacturer Name String [R] (0x2A29)
- Firmware Revision String [R] (0x2A26)
SMP Service (8d53dc1d-1db7-4cd3-868b-8a527460aa84)
- SMP Characteristic [N WNR] (da2e7828-fbce-4e01-ae9e-261174997c48)
   Client Characteristic Configuration (0x2902)
D	11:29:06.233	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
I	11:29:06.300	Connection parameters updated (interval: 50.0ms, latency: 0, timeout: 5000ms)
I	11:29:09.949	Connection parameters updated (interval: 50.0ms, latency: 0, timeout: 500ms)

```

Qpczyk, 2021-02-11:
nRF Connect Manager seems to work fine.

Qpczyk, 2021-03-03:
Is something planned to be done with this issue?

philips77, 2021-03-03:
Well.. I wish I could go back to nRF Connect, but I'm busy with some other things for some time.

jwouters91, 2021-07-15:
Facing the same issue here. I am on the same android, nrf connect and zephyr version. It looks as if the device is not re-booting into bootloader mode prior to where the image confirmation is supposed to take place. Any help would be much appreciated. Thanks!

philips77, 2022-03-16:
The latest mcumgr lib is now available in nRF Connect 4.25.1. Hopefully this has fixed the issue.

closed 2022-03-16 by philips77