nordicsemi/Android-nRF-Connect #145: Reliable write status after disconnection

Ga-ryo, 2021-04-21:
**Describe the bug**
When the connection closes during reliable writing, internal status(?) will not be cleared.

**To Reproduce**
0. Start advertising.
1. Connect to a peripheral which exposes writable characteristics.
2. Tap Begin in "Reliable write". (nRF connect)
3. Close the connection. ( Tapping cross mark and reboot application )
4. Connect to the same peripheral.
5. Write something to a writable characteristic. (without starting "Reliable write")
6. Read logs in central/peripheral. (It shows that the central sends not ATT_OP_WRITE_REQ/ATT_OP_WRITE_CMD but ATT_OP_PREP_WRITE_REQ)
(7. Try 5~6 again. Then nRF connect shows "Error 7 (0x7): GATT INVALID OFFSET", maybe because it doesn't increment offset during first attempt.)

Peripheral side log of 5~7 ↓
```
$ sudo hcidump
HCI sniffer - Bluetooth packet analyzer ver 5.55
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
> ACL data: handle 64 flags 0x02 dlen 10
    ATT: Prepare Write req (0x16)
      attr handle 0x000c, value offset 0x0000
      part attr value  0x00
< ACL data: handle 64 flags 0x00 dlen 10
    ATT: Prepare Write resp (0x17)
      attr handle 0x000c, value offset 0x0000
      part attr value  0x00
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
> ACL data: handle 64 flags 0x02 dlen 10
    ATT: Prepare Write req (0x16)
      attr handle 0x000c, value offset 0x0000
      part attr value  0x11
< ACL data: handle 64 flags 0x00 dlen 9
    ATT: Error (0x01)
      Error: Invalid offset (7)
      Prepare Write req (0x16) on handle 0x000c
> HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
```

**Expected behavior**
At step 5~6, sending ATT_OP_WRITE_REQ/ATT_OP_WRITE_CMD is expected.
Clearing transaction status will resolve this issue.

**Versions (please complete the following information):**
 - Android 11 + nRF connect latest(4.24.3) : Central
 - Linux VM on mac(implemented by noble) + USB BLE dongle(Broadcom) : Peripheral

philips77, 2021-08-24:
The issue is on Android side. It does not clear its state. But I managed to workaround it by always calling `gatt.abortReliableWrite()` just after connection. 
I get `onReliableWriteCompled(gatt, status)` with `status = 133`, but the following write operation seems to be ATT_OP_WRITE_REQ as it should.
I'd say that's a hack and I hope it will work on older phones as well. I tested on Pixel 5 with Android 11.

philips77, 2022-03-16:
This is now release in 4.25.1. The app could be always calling `gatt.abortREliableWrite()` on connection.

closed 2022-03-16 by philips77

philips77, 2022-03-17:
I'm preparing 4.25.2 where aborting reliable write on connection will be optional and initially disabled. To enable, open Settings -> Connectivity. 
The reason is, that some devices don't reply with Not Supported or Success to Abort Execute Write Request and the app just waits there.
As Reliable Write is quite rare to be used, we decided to disable this by default.