nordicsemi/Android-nRF-Connect #148: App gives links to nonexisting bluetooth.org URLs

eugmes, 2021-05-12:
**Describe the bug**
The app gives some links to bleutooth.org pages that no longer exists.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the app
2. Locate any device and click "RAW"
3. Observe the URL on the bottom of the shown dialog.

**Expected behavior**
Only working URLs to bluetooth.com or similar are shown.

**Screenshots**
[image]

**Versions (please complete the following information):**
 - Android version: 9
 - Phone model Nokia 9
 - App Version latest?

philips77, 2021-05-12:
Hello,
Yes, we're aware of this issue. Bluetooth SIG have changed addresses and also decided to keep all service info on PDF only (that's in Definitions). It will be fixed in next release.

philips77, 2021-08-24:
Fixed internally.

closed 2021-08-24 by philips77