nordicsemi/Android-nRF-Connect #149: [DFU] Missing Op Code for 'DFU Control Point'

zoltan-mayer, 2021-05-14:
**Describe the bug**
The 'Abort' Op Code (=0x0C) is not available when writing the 'DFU Control Point'.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to the 'BLE Secure DFU Bootloader' example.
2. Open the 'Secure DFU Service'.
3. Tap on the Write button.
4. Observe the scroll down menu of pre-defined Op Codes.

**Expected behavior**
The 'Abort' Op Code is available with value 0x0C.
Alternatively: The user has the secondary option to enter a raw byte array, similarly to unknown characteristics.

**Versions (please complete the following information):**
 - Android version: 9
 - Phone model: Nokia 8
 - App Version: 4.24.3

**Additional context**
The 'Abort' Op Code will reset the bootloader and it will boot into a valid application, if available.

[image]

philips77, 2021-05-20:
This Op Code has been added for DFU protocol after this feature had been implemented in nRF Connect. I'll add it to my backlog.
What you can do for now is a  workaround:
1. In Menu deselect the option "Parse Known Characteristics"
2. Try to writing the characteristic again
3. This time a custom dialog will appear and you'll be able to send any data

philips77, 2021-08-24:
Fixed internally.

closed 2021-08-24 by philips77