nordicsemi/Android-nRF-Connect #152: Question about delays in nRF Connect DFU process

Resultfactory, 2021-06-16:
We are using a nrf52840 to DFU another nrf52 in the same way as the nRF Connected app.
In the nRF Connect logs we can see delays(wait 1000 in the logs) after the services are discovered and after the notifications.
Why is this delay present? 

We have a issue in our DFU process. When we send the instruction to reboot to the bootloader, by writing to the buttonless dfu characteristic. Then sometimes we get a GATT timeout at this point. Usually the device restarts to the bootloader.

Is a delay required after discovery and enabling notifications?

philips77, 2021-06-17:
Hello,

The source code of the DFU library is public, and you can check where and why the delays are there. For example have a look here: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/b5d587bfbfaa5b1eacebce3a8846746a9d3a9793/dfu/src/main/java/no/nordicsemi/android/dfu/DfuBaseService.java#L1287.
This `waitFor` method is called in several other places in from service implementation classes as well.
Basically, the reason why they were added is that an issue was found at this point, and delaying following operations gave the device or the phone more time to process something, and dfu got more stable. Of course they are all "workarounds", as no such should be needed. The delays are very small and should not lead to any timeouts.

Could you paste the logs from your failed upgrade?

Resultfactory, 2021-06-22:
Thanks for the link, I didnt know where to find the sources.
The comments near the waitfor look like device specific fixes.

I will investigate further, test with and without delays and continue [this issue on the nordic devzone](
https://devzone.nordicsemi.com/f/nordic-q-a/76427/gatt-timeout-on-writing-to-buttonless-dfu-characteristic)

closed 2021-06-22 by Resultfactory