nordicsemi/Android-nRF-Connect #154: SecurityException crash during automated tests

Bernstern, 2021-08-17:
**Describe the bug**
nRF Connect crashes after a `java.lang.SecurityException` is raised during an automated test. We haven't found a way to trigger it with a command in the automated tests or a set of steps to trigger it, but after the recent updates we have been seeing this crash more often. 

We submitted feedback w/ the crash through Androids crash feedback. We can update this issue if we find steps to reliably reproduce. 

**To Reproduce**
Steps to reproduce the behavior:
1. Running automated tests in succession
2. Seems to more happen when the disconnect command is invoked

**Expected behavior**
NA

**Screenshots**
NA

**Versions (please complete the following information):**
 - Android version: 8.0.0
 - Phone model: Pixel 2
 - App Version 4.24.3

**Additional context**
From the android crash log:
Source class: android.os.Parcel
Source method: readException
Line number: 1943

The frames in the stack trace with nordic code were:
no.nordicsemi.android.mcp.util.FileHelper.getContentUri(:3)
no.nordicsemi.android.mcp.test.TestService.onHandleIntent(:60)

philips77, 2021-08-18:
Thank you for the report. We're preparing a new release, I'll make sure it's fixed.

philips77, 2022-03-16:
Hello, could you try it on the new 4.25.1 version?

philips77, 2022-08-11:
I'm closing, assuming it's fixed. Also, I'm releasing 4.25.4 which fixed Automated Tests on Android 13.

closed 2022-08-11 by philips77