nordicsemi/Android-nRF-Connect #158: Scanning not working well with Android 10

pablitorun, 2021-12-01:
**Describe the bug**
Scanning does not always work with Android 10.  In previous versions of Android I never noticed any problems

**To Reproduce**
In the SCANNER window initiate a scan.  Often nothing will be reported.  In the case that it does work it will report advertisements seen and the delay between advertisements, but the delay will stop being updated long before the active scanning actually stops.

To recover the scanning behavior to normal functionality I often have to exit the app.  Disable Bluetooth on the device.  Launch the App and enable bluetooth from within the app.

**Versions (please complete the following information):**
 - Android version: 10
 - Phone model Moto G Power 2021
 - App Version 4.24.3

philips77, 2021-12-02:
I think it's the issue with your phone, not the app. Did you try other phone? We have not encountered such issue on any of the phones we use for development and testing. If the Bluetooth needs to be restarted, that indicates that the phone went into unknown territories and needs help.

pablitorun, 2021-12-03:
I have seen it happen now on two phones.  I will keep trying

closed 2022-03-16 by philips77