nordicsemi/Android-nRF-Connect #159: android System Connect

v630140, 2021-12-13:
Hi guys,

I tried to connect to nRF52832 using the app and it was successful.But when I directly use the BLE in the Android system settings to connect to the device, it will not be able to connect.Please refer to the picture below for the situation.

[image]

What can I do to connect directly to the Android system without APP? Thanks.

philips77, 2021-12-14:
Some app is always needed, otherwise system doesn't know what to do with the connection and disconnects. Some devices, like HID (Human Interface Device) or Audio are supported natively, meaning that the Bluetooth Settings or the OS supports such profile, for those you don't need additional app, but just because there already is an app. For other profiles, which are not supported like that, you need an app that could hold and manage the connection.

closed 2021-12-14 by philips77

v630140, 2021-12-15:
Thank you for your answer.