nordicsemi/Android-nRF-Connect #161: Disconnect happens Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST during Automated tests

royan-hub, 2022-03-16:
**Describe the bug**
During automated tests, when executing disconnect，it displays Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST on phone, but it won't happen when I manually disconnect, it also display disconnecting... and gatt.disconnect()

**To Reproduce**
1. Running automated tests successfully
2. when executing disconnect, it displays Error 22 (0x16): GATT CONN TERMINATE LOCAL HOST on phone
3. when repeatedly setup connect and disconnect, the last disconnect will not work on phone

**Versions (please complete the following information):**
 - Android version: [e.g. 9.0.0]
 - Phone model [e.g. 红米10X]
 - App Version [e.g. 4.24.3]

philips77, 2022-03-16:
The error 22 is reported when the phone disconnects, but the disconnection wasn't initiated nay the user (`gatt.disconnect()`). In the latter case the status is 0 (success).

closed 2022-03-16 by philips77

royan-hub, 2022-03-17:
So how can I eliminate this error 22?

royan-hub, 2022-03-17:
We want to automated test is the same as manual test, 
for automated test, our xml :
[image]
and when we execute the xml file, the phone display:
[image]