nordicsemi/Android-nRF-Connect #162: Android 4.25.0 release is slow and not functional on Android 9 & 10

patrickniessenmakezens, 2022-03-16:
**Describe the bug**
After update to version 4.25.0 connecting to devices is difficult.

**To Reproduce**
Press the scan button and try to connect to a known working device

**Expected behavior**
Connection established and characteristics visible

**Versions used:**
 - Android version: 9 &  phone model Samsung S8
 - Android version: 10 &  phone model Samsung S9
 - App Version 4.25.0

**Additional context**
Also when scanning Beacons their rssi interval is very irregular. Typical Beacon transmission was 97mSec with the older version of the app, now the interval is 4 seconds or larger.

Version 4.21.1 ( on Android 8) works correct and can read these devices

philips77, 2022-03-16:
Thank you for creating the issue. I'll test the app tomorrow. We have S8 in the office.

rabarberski, 2022-03-21:
I upgraded a Pixel4a (Android 11) from 4.24.3 to 4.25.1.
I experience the same issues: 
- app is slow/unresponsive, especially when trying to connect. Android OS warning "nrf Connect isn't responding" pops up.
- connecting not successful so far ("Connecting..." keeps spinning). 
- the log feature (left-to-right swipe) remains empty even on DEBUG level.

nrf Connect 4.24.3 seems to work fine on another Pixel4a (Android 11) that still has the older app version
nrf Connect 4.25.1 seems to work fine on Moto G7 (Android 10).
nrf Connect 4.25.1 seems to work fine on Samsung Galaxy A41 (Android 11), but feels more slow.

philips77, 2022-03-21:
I'm working on a bug fixing release.

rabarberski, 2022-03-21:
**Update**. 
- I've tested some more, also on other similar Pixel4a (Android 11) phones I could get a hold off. They worked fine.
- Looking into the original Pixel4a that was giving issues, I noticed that under App Info it said that it used 12GB internal storage or so (could have been 1.2GB as well, not sure). So, I removed the nrf App and reinstalled it, and also rebooted the phone. Worked fine after that !! 
- So maybe certain App cache/storage  things don't get clear up correctly?

patrickniessenmakezens, 2022-06-08:
This can be closed.

It was related to a specific artifact on our device.
One characteristic intended for debug that was not in the release code though it did cause the nRF Connect app to trigger strange behaviour. Devices using RTT crashed. Non RTT device the nRF Connect app malfunctioned.