nordicsemi/Android-nRF-Connect #163: Cannot make a DFU with McuMgr using the new 4.25.1 version

alexemdesgagne, 2022-03-17:
**Describe the bug**
I'm trying to make a DFU of an app built with MCU Boot and SMP server. The new firmware seems to be "pushed" correctly to the device, but the device do not reset and I don't see the "validating" message either. I don't see the "Application has been sent successfully" toast at the end of the operation. I tried reinstalling the previous version (4.24.3) and all the DFU procedure worked properly. 

Like I said, I really think the firmware is correctly "pushed" to the device because when I'm trying the update a second time, I'm not seeing the progression graph. I think it may be an error during the image swap.

**To Reproduce**
To reproduce this issue, you must have an SMP server app flashed on a device.
Steps to reproduce the behavior:
1. Go to nRF Connect for Mobile
2. Click on your device
3. Click on "DFU" logo in the top-right corner.
4. Choose the app_update.bin file to use.
5. You should see the progression bar increasing.

**Expected behavior**
We should be able to update a firmware using nRF Connect like in the previous version (4.24.3).

**Screenshots**
Here's the log when performing the DFU. There's no error.
https://imgur.com/a/mXlH4qB

**Versions (please complete the following information):**
 - Android version: [e.g. 11.0.0]
 - Phone model [e.g. Moto Z3 Play]
 - App Version [e.g. 4.25.1]

philips77, 2022-03-17:
Thank you for creating the issue. I'll look into that tomorrow.

jheiskan, 2022-03-22:
@philips77 We have same issue, reset and reboot is not executed.

philips77, 2022-03-22:
Hi, could you go to Settings -> MCU manager options -> disable "Erase application Storage" and try again?
Thank you!

alexemdesgagne, 2022-03-22:
@philips77 That worked for me! 🎉

jheiskan, 2022-03-23:
@philips77 Works also for us. I wonder why this Erase application storage option is not turned OFF by default in the new application? That would be better for app backward compatibility too.

philips77, 2022-03-23:
In the next version the app will still have it enabled by default, but will not abort if Not Supported is received.
This feature was added in NCS 1.8, it allows to remove app settings on update, and is useful when you're switching to a different app or upgrading to a new major version.

closed 2022-03-25 by philips77