nordicsemi/Android-nRF-Connect #172: Did any Sniffer tool will helps in Analyzing packets from BLE- Mesh node to Mobile App

Dharani-1195, 2022-05-30:
We are working towards the Nordic Bl653-dvk Will any sniffer tool is available to help in analyzing packets from BLE- Mesh node to Mobile App.

- Testing BLE-Mesh network.

Please kindly suggest or provide the links for testing for analyzing the packets.

philips77, 2022-05-30:
Hi,
If you're using our mobile mesh libraries ([iOS](https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library), [Android](https://github.com/NordicSemiconductor/Android-nRF-Mesh-Library)) they print bunch of data in the logger. The iOS library allows to connect a custom logger delegate, and sends logs with different categories matching layers. The Android one prints to LogCat.

If you want a sniffer, the Ellisys sniffer we have provides option to decrypt mesh messages, and is a great debugging tool. Highly recommend, but I guess it's very expensive.

It is possible to do a mobile sniffer for mesh on Android for the ADV Bearer, as the API allows to get raw data from the scanned package. We have such task in our backlog.

Dharani-1195, 2022-05-30:
> 
Thank you, Phillips.

closed 2022-05-31 by philips77