nordicsemi/Android-nRF-Connect #173: No code changes for v4.25.x releases?

rabarberski, 2022-06-07:
There have been a couple of new releases for the app, i.e. v4.25.0, v4.25.2 and v4.25.3. 
The are available in github under releases, https://github.com/NordicSemiconductor/Android-nRF-Connect/releases

I wanted to look up detailed code changes in some v4.25.x releases.
However, github doesn't show any code changes in the commits, neither for the master branch nor e.g. the 4.25.3 branch (https://github.com/NordicSemiconductor/Android-nRF-Connect/tree/version_4.25.3). Oldest commit is f10b32b from 25 Sep 2020.
Is the git repo itself not updated with the new code?

philips77, 2022-06-07:
Hello,
This repo is just for documentation and issue tracker. The source of the app is not open and we keep it internally.
What changes are you interested in? Perhaps I can help you.

rabarberski, 2022-06-07:
> This repo is just for documentation and issue tracker. The source of the app is not open and we keep it internally.

Ok, fair enough. It seems I looked over the remark in the repo readme saying "_Note: The source code of nRF Connect is not available._". Perhaps it could be moved to the top to be more visible, and state the version of nrfConnect it was frozen at?

>What changes are you interested in? Perhaps I can help you.

I was wondering if the way that characteristic byte arrays  are sent over, has changed in v4.25, compared to v4.24.
Our BLE device exposes a writable 4-byte characteristic, which we program using the nrf Connect app. Previously, we entered the hex value to be written as e.g. 1234 (i.e., a 2-byte hex value 0x1234 = 4460 integer) in the byte array field of the nRF app, and it was processed by our device 'correctly' as 00 00 12 34.
Now, we need to enter all the leading zero's as well (i.e. 00 00 12 34) , otherwise we end up with 0x12340000.
It could also be our BLE device  subcontractor has changed something in the byte handling; still to be checked/confirmed.

philips77, 2022-08-09:
No, seems like this change originates in your firmware. nRF Connect just sends 2 bytes, and it's up to your device to convert them to UInt32. You may also use UInt32 field in the dialog box, which will sent them in Little or Big Endian.

closed 2022-08-09 by philips77