nordicsemi/Android-nRF-Connect #174: Decoding of CGM Measurement Packet is Incorrect

dfridley-roche, 2022-07-02:
**Describe the bug**
Decoding of CGM Measurement Packet is incorrect.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to a CGMS device that is sending CGM Measurement packets with annunciations and one or more optional fields.
2. Subscribe to CGM Measurements

**Expected behavior**
Decoding of the packet should show annunciation flags set and correct Trend, Quality, E2E-CRC values

For example.
Data packet:
`0D-43-73-00-2C-01-03-24-00-1A-00-3E-04`

Should decode as:
```
Glucose concentration: 115.0 mg/dL
Sequence number: 300 (Time Offset in min)
 Cal/Temp Info:
- Time Synchronization Required
- Calibration Not Allowed
Trend: 36.0 mg/dL/min
Quality: 26.0%
E2E-CRC: 0x043E
```
**Screenshots**
Incorrect decoding of CGM Measurement Packet
[image]

**Versions (please complete the following information):**
 - Android version: 12
 - Phone model Pixel 5
 - App Version 4.25.3

**Additional context**
See Bluetooth 
[Gatt Specification Supplement 6](https://www.bluetooth.com/specifications/specs/gatt-specification-supplement-6/) section _CGM Measurment_

philips77, 2022-07-14:
Thanks! We'll fix it in the next release.

philips77, 2022-08-09:
Fixed internally.

closed 2022-08-09 by philips77