nordicsemi/Android-nRF-Connect #185: Crash app when log file is big

omidxs, 2022-12-30:
Hi
I used a device to get data from an object. I send data with BLE and receive them with “nRF Connect”. I need to process data with MATLAB so by using “nRF logger” I save the log file (.txt) and extract data from it. When I want to save small data (like 30 minutes measurements) it works properly. But if I want to save a big file (8-hour measurements), the app crashes. I think this app can’t create large-size files. 
The .txt file of 30 minutes measurements is about 12 MB.

Hint: My phone’s CPU is S778G and has 8G RAM and storage are enough.

I think my problem is same with #83 :
https://github.com/NordicSemiconductor/Android-nRF-Connect/issues/83

**Versions **
 - Android version: [12]
 - Phone model [xiaomi 11 lite 5g ne]
 - nRF logger Version [1.9.1]
 - nRF connect Version [4.26.0]

fproko, 2023-04-06:
I have the same problem. 

- Android version: [11]
- nRF connect Version [4.26.1]

fproko, 2023-04-08:
Where is the app's database? I have routed phone.

vhakulinen, 2024-05-03:
I had some trouble exporting large (~50mb) log file on my phone. No matter where I tried t he export (the list view, or from the log view), the exported file ended up being empty. But when exporting from the log view, the screen would go black for a while and the app would soon restart.

What ended up working, was locking the screen once the screen went black. Maybe the system was killing the app because it was not responding, but the lock screen altered this default functionality.

Mayitzin, 2024-12-16:
> What ended up working, was locking the screen once the screen went black. Maybe the system was killing the app because it was not responding, but the lock screen altered this default functionality.

I find it incredibly simple, but this also worked for me.

1. I saved all files I could the normal way.
2. When a file could not be saved, I just start the saving process like normal, and lock my phone when the screen goes black.
3. Seconds later I unlock the phone, and it notifies me the file has been saved.
4. I check the downloads folder (they're saved there), and the entire file is saved.

I don't know what the log file size limit is, but my problematic files are around 42 MB each.

philips77, 2024-12-17:
The latest update (4.29.1) MIGHT improve this a bit by moving saving to another thread, so hopefully the screen won't get black... Please try.