nordicsemi/Android-nRF-Connect #186: Bonded device does not recognise new service/characteristic UUIDs

mgaman, 2023-01-08:
I am using nRF Connect and Arduino ESP32_BLE library to learn Bluetooth
Essentially I program an ESP32 as a server and send services. The nRF connect is used as the client to display the contents of the service.

Once my ESP32 has sent a service and the nRF has received the service , nRF forever remembers the structure of the original service. If I change the service e.g. add another Characteristic, it never gets recognised.

To bypass the problem I have to unpair the  device, reprogram with a name not previously used, re-pair to my phone and then open the client under its new name. Now I see the additional characteristic.

I also tried going to Android settings/Apps/Manage/nRF and deleting data but to no avail.
I also tried on the Bonded menu/Delete Bond Information, also to no avail (BTW that is it supposed to do?)

**To Reproduce**
Go to https://github.com/nkolban/ESP32_BLE_Arduino and run the BLE_notify example (I modified the UUIDs ans payload to make it appear to be a Heart Rate Monitor but that is beside the point).
Once you see the data appearing on nRF Connect modify the code to add a 2nd characteristic. DO NOT change the device name in BLEDevice::init(). Open nRF, go ro Bonded list and Connect the device. nRF will not show the new characteristic.
Unpair the device from your phone. Change BLEDevice::init() with a new name e.g. instead of ESP32 make it ESP32x. Pair ESP32x to your phone and it will appear in the Bonded list of nRF. Connect and the Client screen will now show the new characteristic.

**Versions (please complete the following information):**
 - Android version: [e.g. 10 QKQ1.191014.001]
 - Phone model [Xiaomi Redmi 8]
 - App Version [e.g. 4.26.0]

philips77, 2023-01-09:
To refresh services go to top-right menu on the connection tab and select "Refresh services".

philips77, 2023-01-09:
Caching service is done by the Android system, not the app. How it caches changes from version to version. In theory it should cache services of bonded devices until you unbind them, or a Service Changed indication is received from it, and should cache services of not bonded devices if they don't have Service Changed characteristic (which indicates, that the services will never change). If they do, the OS should never cache services as they may change in every connection (in theory).
I'm not sure how the new Database Hash characteristic works.
Anyway, Android has never done caching properly. In some versions it caches everything, in some nothing, in some only for bonded. But it allows to call hidden `refresh()` method to clear the cache manually, and that is what nRF Connect does on request.

closed 2023-01-09 by philips77

AdemYavuzCelik, 2024-08-12:
You are able to call hidden refresh() method without a connection in your application. Could you please let me know how you are doing this? @philips77

philips77, 2024-10-10:
Hi,
This method is called on `BluetoothGatt` object. To obtain an instance of it you need to call `connectGatt(...)`, but then you can disconnect from the device and clear cache before calling `close()`. This just clears the internal cache on the phone. If the connection is active, the call will also trigger new service discovery.