nordicsemi/Android-nRF-Connect #191: Element 'assert-server-service' does not have a match in class no.nordicsemi.android.mcp.test.domain.Test at line 61

kanagala388, 2023-03-20:
I have Apple Notification center service configured in nrf app on my android device. I want to send some notifications to my remote device.
Connection and bonding are working. But after connection "assert-server-service" element throwing the following error.

**Parsing test suite...FAIL
Element 'assert-server-service' does not have a match in class no.nordicsemi.android.mcp.test.domain.Test at line 61**

Steps to reproduce the behavior:
1. After connection use assert-server-service to verify the service existence. Please refer to the below xml snippet for better understanding

Sample XML snippet:
`<connect/>
		<sleep timeout="10000" />
		<assert-server-service description="Learning" instance-id="0" uuid="7905f431-b5ce-4e99-a40f-4b1e122d00d0">
			<assert-characteristic description="Ensure 09fbf120d-6301-42d9-8c58-25e699a21dbd characteristic" uuid="9fbf120d-6301-42d9-8c58-25e699a21dbd">
				<property name="NOTIFY"/>
			</assert-characteristic>
		</assert-server-service>
		<send-notification description="Send 0x0015060100000000 as notification from 7905f431-b5ce-4e99-a40f-4b1e122d00d0" characteristic-uuid="9fbf120d-6301-42d9-8c58-25e699a21dbd" service-uuid="7905f431-b5ce-4e99-a40f-4b1e122d00d0" value="0015060100000000"/>`

**Versions (please complete the following information):**
 - Android version: 10
 - Phone model: Oppo
 - App Version: 4.26.0

**Additional context**
Add any other context about the problem here.

philips77, 2023-03-21:
Looks like you're trying to use tags for Macros in Automated Tests. Currently tests don't support GATT server, just the client.

Here's the documentation for asserting client services: https://github.com/NordicSemiconductor/Android-nRF-Connect/tree/main/documentation/Automated%20tests#assert-services-and-characteristics

Macros support both client and server and can also assert services on the phone: https://github.com/NordicSemiconductor/Android-nRF-Connect/tree/main/documentation/Macros#service-assertion

philips77, 2023-03-21:
You may ask why Automated tests do not support server... this is just due to lack of resources. At some point it will be implemented, I hope. Sorry for that.

closed 2023-03-21 by philips77