nordicsemi/Android-nRF-Connect #194: test.bat script fails to push file on Android 13

udth, 2023-06-16:
**Describe the bug**
I have tried to use the bat file provided in:
https://github.com/NordicSemiconductor/Android-nRF-Connect/blob/main/documentation/Automated%20tests/DFU%20Tests/test.bat 

The script was attempted with an Pixel phone running Android 13, wit USB debugging enabled.

The test script did not proceed beyond:
call adb %S_DEVICE% push %XML_PATH% "/sdcard/Android/data/no.nordicsemi.android.mcp/files/Test/%XML_FILE%" > nul 2>&1

As a workaround I tried creating a new folder on the phone, and replacing "/sdcard/Android/data/no.nordicsemi.android.mcp/files/Test/%XML_FILE%" to "/sdcard/tmp/%XML_FILE%"

With this change I was able to successfully get beyond the push step, but the nRF Connect app on the phone popped up a notification saying "Test suite file not found."

**To Reproduce**
Steps to reproduce the behavior:
1. Download the test.bat script from https://github.com/NordicSemiconductor/Android-nRF-Connect/blob/main/documentation/Automated%20tests/DFU%20Tests/test.bat and "https://github.com/NordicSemiconductor/Android-nRF-Connect/blob/main/documentation/Automated%20tests/sample_test_scanning.xml"
2. Connect a USB debug enabled phone running Android 13 and execute the test with ".\test.bat -D [DeviceID] .\sample_test_scanning.xml"
3. Observe that the test.bat script does not proceed beyond trying to push the file.
4. Create a folder "/sdcard/tmp/" on the phone.
5. Edit the test.bat file replacing all occurrences of "/sdcard/Android/data/no.nordicsemi.android.mcp/files/Test" with "/sdcard/tmp/"
6. execute the test agina dn observe error on the phone.

**Expected behavior**
The test script is successfully executed.

**Screenshots**
[image]

**Versions:**
 - Android version: 13
 - Phone model: Pixel 7 Pro
 - App Version: 4.26.1

philips77, 2024-01-15:
Please, try nRF Connect version 4.27.1. 
We added a script *test.sh* for Mac/Unix and it works on our side. Tried on Android 14 and the file is pushed successfully, and the result is pulled successfully.

> With this change I was able to successfully get beyond the push step, but the nRF Connect app on the phone popped up a notification saying "Test suite file not found."

nRF Connect app does not have permission to access your fils system. It can only see its internal folders, that's why there path is */sdcard/Android/data/no.nordicsemi.android.mcp/files/Test*.

philips77, 2024-01-15:
Also, sorry for half a year delay!

closed 2024-01-15 by philips77