nordicsemi/Android-nRF-Connect #198: Compatible with permissions of Android 12 and above

hjjice, 2023-08-30:
Why does the APP still apply for location permissions on versions above Android 12, which seems to be judged to be excessively asking for permissions

philips77, 2023-09-03:
Hi, this is a great question. Let me answer.

Since Android 12, as you know, a new set of permissions were added: 
* `BLUETOOTH_SCAN`, 
* `BLUETOOTH_CONNECT` 
* `BLUETOOTH_ADVERTISE`. 

The new permissions replaced the previous ones: `BLUETOOTH` and `BLUETOOTH_ADMIN`. 
More about that here: https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

A new feature related to scanning for Bluetooth LE advertising packets allows to add `neverForLocation` modifier:
```xml
<manifest>
    <!-- Include "neverForLocation" only if you can strongly assert that
         your app never derives physical location from Bluetooth scan results. -->
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"
                     android:usesPermissionFlags="neverForLocation" />
    ...
</manifest>
```
The problem with this flag is that it excludes iBeacons and Eddystone from the scan results. Many of the users of nRF Connect do need to scan for them for development purposes, so that was no an option.

Initially, we thought that granting `BLUETOOTH_SCAN` without `FINE_ACCESS_LOCATION` would let the app scan all packets except from iBeacons and Eddystone, as if `neverForLocation` was set. That would be perfect, as users could decide: beacons or no beacons. But no, without `FINE_ACCESS_LOCATION` we get nothing. We either have to resign from scanning for beacons at compile time (adding the manifest flag), or ask the user for location. To make it even harder, now it has to be the `FINE_ACCESS_LOCATION`, which we're requesting, but users can downgrade it to `COARSE_ACCESS_LOCATION` ("ok, you need my location, so at least don't have precise one"), but this also doesn't work and we need to ask again for the fine one.

The only thing we can do is not use the location in any way. That's also one of the reasons nRF Connect doesn't have `INTERNET` permission. Even if we'd collect the location, we can't send it anywhere. We respect users' privacy and didn't use the Internet or any form of analytics in any version of the app, wanting the app to be trusted and safe to use even in projects that are not public.

However, we may start using Internet in some future versions if we find it necessary for new features. Still, old versions of the app are available here on GitHub in *releases*.

philips77, 2023-09-03:
And yes, we're aware that requesting location isn't only about the permission to the app, but also enabling Location services globally on the device, which users may not want. But location can be disabled per-app, so this can be work-around by denying it to all other apps. Yes, very bad user experience just to allow scanning for beacons...

Note, that on iOS beacons (only iBeacons) are always excluded from Bluetooth LE scan results. There is a separate API to access them, using *Core Location* instead of *Core Bluetooth*. That means that Android is the only (mobile) option for developers working with this technology.

closed 2024-01-09 by philips77