nordicsemi/Android-nRF-Connect #200: Does the nRF-Connect app use the AndroidRxBle lib?

MaxWebfactor, 2023-10-05:
Hi there,
I am currently developing an application based on the RxAndroidBle2 library to establish a connection. It works pretty well up to a point where we want to discover about 425 handles. Unfortunately the library fails with a timeout after 20 seconds. 424 handles work fine. 

A colleague rebuilt the connection setup with the nRF-Connect app on Android with the same result as described above. 

So my question is Does the nRF-Connect app use the AndroidRxBle lib?
I am just asking to know if this is an Android issue or more of a data flow issue within the app.

philips77, 2023-10-05:
Hi,
We're not using RxAndroidBle2 library. This must be Android limitation. Do you have all characteristics in a single service, or in multiple?

MaxWebfactor, 2023-10-05:
Hi,
there are multiple services. In one configuration, I have to discover about 39 characteristics in one service. Maybe that's where the discovery breaks on Android. There was an issue in 2018 where 36 characteristics per service was a limitation.

But according to your comment, it appears that the issue arises from the Android OS rather than a specific library.

Thanks for your response. 
I would appreciate your thoughts on our problem.